ProteoWizard/pwiz #2625: Validation error in mzML with Thermo RAW from MAT95XP

tilfischer, 2023-06-20:
Dear all,

We have Thermo .RAW data from a Thermo Finnigan MAT95XP MS (double focused sector mass spectrometer with BE geometry with EI/FAB ion source), which can be converted with msconvert. However, this does not give valid files, if validated with OpenMS/FileInfo 2.8.0.

File are available here: https://drive.google.com/drive/folders/1TJ79i1axOfTh7qO8qC5gHLFtAmzeD10H?usp=sharing

Validation of MAT95XP-File001.mzML with FileInfo gave:

```
-- General information --

File name: MAT95XP-File001.mzML
File type: mzML

Validating mzML file against XML schema version 1.1.0
Validation error in file 'MAT95XP-File001.mzML' line 40 column 25: missing elements in content model '(source+,analyzer+,detector+)'
Failed - errors are listed above!

Semantic validation is not performed due to previous errors!
FileInfo took 0.02 s (wall), 0.00 s (CPU), 0.00 s (system), 0.00 s (user); Peak Memory Usage: 31 MB.
```

The componentList tag in the generated mzML is:

```
       <componentList count="1">
          <source order="1">
            <cvParam cvRef="MS" accession="MS:1000389" name="electron ionization" value=""/>
          </source>
        </componentList>
```

I manually fixed the error by substitution of the componentList tag with the following lines:

```
<componentList count="3">
        <source order="1">
          <cvParam cvRef="MS" accession="MS:1000389" value="" name="electron ionization" />
        </source>
        <analyzer order="2">
          <cvParam cvRef="MS" accession="MS:1000443" value="" name="mass analyzer type" />
        </analyzer>
        <detector order="3">
          <cvParam cvRef="MS" accession="MS:1000624" value="" name="inductive detector" />
        </detector>
</componentList>
```

Validation with FileInfo gave a valid file, with only one warning by the semantic validation:

```
-- General information --

File name: MAT95XP-File001-fixed.mzML
File type: mzML

Validating mzML file against XML schema version 1.1.0
Success - the file is valid!

Semantically validating mzML file:
Warning: Unit CV term used, but not allowed: UO:0000186 - dimensionless unit of term MS:1000786 - non-standard data array
Success - the file is semantically valid!
FileInfo took 0.08 s (wall), 0.05 s (CPU), 0.00 s (system), 0.05 s (user); Peak Memory Usage: 39 MB.
```

It seems that the analyser is not specified in the .RAW provided, hence, msconvert does not add this information to the mzML. However, the schema always requires the triplet of source, analyser and detector, hence, FileInfo throws an error on validation. 

We also have other .RAW from a MAT95XP hyphenated with a Finnigan Trace GC Ultra. This .RAWs show the same behaviour. 

The providers of the file linked above provide them under a CC0 license i.e. the files could be added the the [test data](https://github.com/ProteoWizard/pwiz/tree/e557427743107973c6fff118500f38e3a4aec69e/pwiz/data/vendor_readers/Thermo/Reader_Thermo_Test.data).

Best and thank you for fixing this issue,
Tillmann

chambm, 2023-06-21:
Thanks for providing the test file. That MAT95XP file doesn't have its instrument model or instrument name parameter set (they're blank strings), so there's nothing for ProteoWizard to key off of to determine the instrument. The file has "MAT95" in the header comment, but AFAIK that's totally free text set by the user/method, so not reliable for determining model. Hmm, it does have that info in the status log though. Maybe I can add a check in that:
[image]

Even if I can't get the instrument model though I should probably drop the componentList entirely so as not to generate an invalid document.

tilfischer, 2023-06-22:
Dear Matt, 

while I am not that familiar with the actual Thermo RAW format itself, I also saw this information in the status log via Xcalibur. Furthermore, the instrument software does not only generate one .RAW after reach run but several of them. The MAT95-File001.RAW is the actual raw data, while further data such as calibration data is saved in other files ending with "-c1", "-c2", "-c3"... 

I uploaded such a file including its mzMLs to the google drive linked above. The status log of such a .RAW provides other information than the one above:

[image]

Best and thank you,
Tillmann

sneumann, 2023-06-23:
Hi, IMHO there are several options to move forward:
1) leave-as-is and document the behavior. The resulting mzML files will most likely be read by the majority of tools out there, but validation with OpenMS will bark on the incomplete `componentList`.
2) Drop the entire `<componentList>` if incomplete, this will fix the validation error. However, that means loosing the ionisation, which to me is the most valuable piece of information. 
3) In the absence of actual information, use the rather uninformative cvTerms MS:1000443 "mass analyzer type" and MS_1000026 "detector type". With these values the mzML validation will succeed.
4) Add tons of special casing for this instrument. 

Given that the instrument is from the 1990s, I'd be hesitant to ask for 4). But I think providing valid files should be an important goal, so I'd favour 3) (plus documenting the behaviour) over 1) and 2). But I wouldn't complain if 4) is  easy enough to implement :-) 

Regarding the calibration data, the `"-c1", "-c2", "-c3"...` files described by Tillmann remind me of how Waters puts different functions into different files, only that here the files are not placed into their own directory. 

Yours, Steffen

PR #2644 (2023-07-03): - fixed reading instrument model for very old Thermo RAW files
- fixed reading instrument model for very old Thermo RAW files which may store instrument model only in status log
- added commandline parameters for msconvert to output file

Closes #2625
closed 2023-07-03 by chambm