ProteoWizard/pwiz #2626: Cannot convert from continuous to centroid with vendor peakpicking

PierreSnell, 2023-06-20:
When dealing with QCP and QCI files, we have a setup that returns continuous data. 

I would like to use msconvert to get centroid so it would be the same as all the rest of the data we have. 

I'm currently using : 

```python 
    arguments = ""
    arguments += f"wine msconvert {docker_file} "
    arguments += "--outdir /out_data --mzML "

    if "_QCP_" in docker_file.upper() or "_QCI_" in docker_file.upper():
        arguments += """--filter "peakPicking true 1-" """

    # # Tried with all the arguments bellow but removed them to pinpoint the issue.    
    # arguments += f"""--filter "lockmassRefiner mz={lock_mass} tol=0.5" """
    # arguments += """--filter "msLevel 1" """
    # arguments += """--filter "zeroSamples removeExtra" """
    # arguments += "--zlib "  
    # arguments += "--numpressAll "
    # arguments += "--32 "  # set default binary encoding to 32-bit precision
    # arguments += "--inten32 "  # encode intensity values in 32-bit precision
    # arguments += "--mz32 "  # encode m/z values in 32-bit precision
```

However I got a
```
2023-06-20T20:04:57.135427847Z format: mzML  
2023-06-20T20:04:57.135481769Z     m/z: Compression-None, 64-bit
2023-06-20T20:04:57.135485667Z     intensity: Compression-None, 32-bit
2023-06-20T20:04:57.135488582Z     rt: Compression-None, 64-bit               
2023-06-20T20:04:57.135491097Z ByteOrder_LittleEndian                            
2023-06-20T20:04:57.135493662Z  indexed="true"                                           
2023-06-20T20:04:57.135496327Z outputPath: /out_data
2023-06-20T20:04:57.135498781Z extension: .mzML                                               
2023-06-20T20:04:57.135501206Z contactFilename: 
2023-06-20T20:04:57.135503591Z runIndexSet:                                                    
2023-06-20T20:04:57.135508370Z spectrum list filters:
2023-06-20T20:04:57.135510794Z   peakPicking true 1-
2023-06-20T20:04:57.135513149Z   msLevel 1
2023-06-20T20:04:57.135518318Z chromatogram list filters
2023-06-20T20:04:57.135523148Z filenames:  MySuperFile.raw
2023-06-20T20:04:57.135636652Z processing file: MySuperFile.raw
2023-06-20T20:04:58.637570736Z calculating source file checksums
2023-06-20T20:05:50.877500090Z writing output file: MyOutFile.mzML
2023-06-20T20:10:39.740727767Z Error writing run 1:
2023-06-20T20:10:39.740760378Z File Open Error
2023-06-20T20:10:39.741977092Z Conversion failed for 1 runs in  MySuperFile.raw
2023-06-20T20:10:39.744486502Z Error processing file  MySuperFile.raw
```

I tried all the combinations of parameters, on many different files from different days/batches ,without success.... 

Thanks in advance, 
Have a great day.

closed 2023-06-21 by PierreSnell

chambm, 2023-06-21:
Peak picking on Waters .raw can be pretty finicky. Also the Waters peak picker caches the centroid data inside the .raw directory, so it needs write access. Was that the issue?

PierreSnell, 2023-06-21:
Hello @chambm, 

I don't really know to be honest, after hours of switching the filters it seemed to black-magically work
Maybe was the ``--filter "peakPicking true 1-"` that I changed for explicit `--filter "peakPicking true msLevel=1"` 
So I will keep it like that (thus closed the issue)

Thanks a lot for the fast reply ! 
Have a great day.