ProteoWizard/pwiz #2659: MSConvert idling

bernt-matthias, 2023-07-13:
I have a (Galaxy) job (195680 for my reference) that is idling around after an initial short phase of 100% CPU usage after I see some messages like:

```
[SpectrumWorkerThreads::work] error in thread: [References::resolve()] Failed to resolve reference.
  object type: struct pwiz::msdata::DataProcessing
  reference id: dp_sp_8
  referent list: 1
    dp_sp_0
```

Any idea what we could try? Would it be an idea to update the Galaxy tool to the latest container?

bernt-matthias, 2023-07-13:
Indeed, it is `mzML` (`indexedmzML`).

bernt-matthias, 2023-07-14:
Thanks for the feedback. Seems that the users tried to merge mzml files with a self made script.

closed 2023-07-14 by bernt-matthias

chambm, 2023-07-14:
FYI msconvert does have a --merge mode for that kind of thing but it's rarely used AFAIK. It does set-difference-based merging of the global referent lists though, i.e. when merging A and B, the output file has the union of the dataProcessing elements from A and B.