ProteoWizard/pwiz #2668: Validation error for indexList with empty indexes

sneumann, 2023-07-19:
Hi, 
when writing out indexed mzML, an empty  `<index>` violates the XSD:
```
  <indexList count="2">
    <index name="spectrum">
      <offset idRef="scan=1">4568</offset>
    </index>
    <index name="chromatogram">
    </index>
  </indexList>
```

In the XSD specification the (implicit) `minOccurs` for `<offset>` is 1:
https://github.com/HUPO-PSI/mzML/blob/81e0145f65dd7abf56a9bea51bfdf66ed7767905/schema/schema_1.1/mzML1.1.3_idx.xsd#L24

```
Validation error in file './neg-MM8-indexed.mzML' 
line 118 column 13: empty content is not valid for content model '(offset+)'
```
Yours, 
Steffen

chambm, 2023-07-19:
That is weird. Are you sure you're validating against 1.1.3_idx? Because allowing minOccurs=0 is the only difference between 1.1.1_idx and 1.1.2_idx:
```
8c8
<                  version="1.1.2">
---
>                  version="1.1.1">
14c14
<           <xs:documentation>Index element containing zero or more offsets for random data access for the entity described in the 'name' attribute.</xs:documentation>
---
>           <xs:documentation>Index element containing one or more offsets for random data access for the entity described in the 'name' attribute.</xs:documentation>
26c26
<       <xs:element minOccurs="0" maxOccurs="unbounded" name="offset" type="dx:OffsetType">
---
>       <xs:element maxOccurs="unbounded" name="offset" type="dx:OffsetType">
```

```
<xs:complexType name="IndexType">
    <xs:sequence>
      <xs:element minOccurs="0" maxOccurs="unbounded" name="offset" type="dx:OffsetType">
        <xs:annotation>
          <xs:documentation>File pointer offset (in bytes) of the element identified by the 'id' attribute.</xs:documentation>
        </xs:annotation>
      </xs:element>
    </xs:sequence>
    <xs:attribute name="name" use="required">
      <xs:annotation>
        <xs:documentation>The name of the entity the index entries are pointing to.</xs:documentation>
      </xs:annotation>
      <xs:simpleType>
        <xs:restriction base="xs:string">
          <xs:enumeration value="spectrum" />
          <xs:enumeration value="chromatogram" />
        </xs:restriction>
      </xs:simpleType>
    </xs:attribute>
  </xs:complexType>
```

sneumann, 2023-07-19:
Ah, my bad, seems the OpenMS FileInfo I used was too old, and I got lost between the XSD versions when double checking. Yours, Steffen

closed 2023-07-19 by sneumann