ProteoWizard/pwiz #2680: 0 Intensities in output mzML

PierreSnell, 2023-08-02:
When converting continuous data from Waters (.raw), or data with Ionic mobility (Waters .raw), the resulting mzML file still contains data points with a mass, a retention time but intensity at 0.

I'm using (python code to run the docker image but all the parameters are here): 

```python
arguments += f"wine msconvert {docker_file} "
arguments += "--outdir /out_data "
arguments += "--mzML "  # write mzML format [default]
arguments += "--zlib "  # use zlib compression for binary data
arguments += "--numpressAll "  # same as --numpressLinear --numpressSlof
arguments += "--32 "  # set default binary encoding to 32-bit precision
arguments += "--inten32 "  # encode intensity values in 32-bit precision
arguments += "--mz32 "  # encode m/z values in 32-bit precision

if "_QCP_" in docker_file.upper() or "_QCI_" in docker_file.upper(): # QC files as they're continuous data not already centroid
    arguments += """--filter "peakPicking true msLevel=1" """

if "_HDMSE_" in docker_file.upper(): # Ionic mobility raw data
    arguments += "--combineIonMobilitySpectra "

arguments += f"""--filter "lockmassRefiner mz={lock_mass} tol=0.5" """
arguments += """--filter "msLevel 1" """
arguments += """--filter "zeroSamples removeExtra" """
``` 

Thus the output mzML data is twice or thrice as big because of those 0s. 

Thanks in advance, 
Have a great daY.

chambm, 2023-08-02:
Continuous (profile) data is expected to have 0s to bound the peak profiles. Otherwise the data would look really weird when plotted and it can mess with peak fitting algorithms as well. If you're talking about the zeroSamples filter, note that it's removing EXTRA zeros, not ALL zeros. I think you can tell it how many flanking zeros to keep and it defaults to 1.

PierreSnell, 2023-08-03:
Hi @chambm, Thanks for the fast response! 

Doing `--filter "peakPicking true msLevel=1"`  I was expecting to get the centroids (from the vendor) and thus not have the bounding zeros. Is it the behaviour I should expect ? Else could I get that ? 

Thanks for the precision about the `zeroSamples` I understood it correctly.

chambm, 2023-08-03:
Well, you are specifying only msLevel=1 to be centroided, so I'd expect centroided MS1s and profile MS2s. Is that what you're saying? Also only the QC files would be getting that peakPicking. If you're getting 0 samples for MS1s with a command that included peakPicking, are the spectra still in profile mode? Or are they 0 samples in a centroided spectrum?

PierreSnell, 2023-08-04:
Sorry if I wasn't clear... 

I'm specifying `--filter "msLevel 1"` for all as I just want MS1 data. 

Yes, only the QC will be peakpicked as it's acquired in a continuous manner. The non-QC data is already acquired in a non-continuous so I don't need peakpicking for them (it's working fine alreay). 

I have the zeros problem only for the QC (continuous acquired) data, I don't have this problem for the non-qc (already vendor peak acquired data). 

However, for the Ion Mobility data, they're non-continuous acquired, so I should have no zeros (like the non-qc non-ion-mobility which are perfectly fine). 
But I got data with a mass, a retention time, _no (0) intensity_, and ionic mobility. 
So I don't know if it comes from the vendor, my understanding of physics (how can I get an ionic mobility without intensity..), or if we should remove them at conversion time. 

I hope I better explained my problem and my use case... 
Thanks a lot for you precious help.

chambm, 2023-08-04:
Can you provide an example .raw file for both cases? I'm not sure I have any non-continuous Waters IMS data.

chambm, 2023-08-07:
Yes I think the latest Docker image should address the problem you saw with the QC files not being centroided. Let me know if it didn't.

PierreSnell, 2023-08-08:
I've tried the latest docker image (version : 
```json
{
    "Id": "sha256:aa3912ac1c59e4f19c39a69bbacd764a17ec1199f48fcd50f374a84036fb7267",
    "RepoTags": ["chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest"],
    "Created": "2023-08-04T18:42:50.856597747Z",
)
```

Here is the code to convert 
```python 
# Run Docker
docker.DockerClient().containers.run(
    "chambm/pwiz-skyline-i-agree-to-the-vendor-licenses",
    _get_conversion_arguments(docker_file), # Parameters on top of this issue
    volumes={
        f"{NAS_ROOT}": {"bind": "/data", "mode": "ro"},
        f"{output_directory}": {"bind": "/out_data", "mode": "rw"},
    },
    stdout=True,
    stderr=True,
    stream=True,
    detach=True,
    remove=True,  # To unmount the volume.
    auto_remove=True,
)
data = load_with_pyopenms_or_pyteomic(mzml_output_by_docker)
np.savez(output_path, data)
```

And the one to load : 
```python
array = np.load(Path("Path/To/My/File.npz"))["arr_0"]
print(array.shape[0])
print(len(array[:,2].nonzero()[0]))

# 166997200
# 122351946 Should be the same as the one above if no zeros. 

print(array)
# array([[   1.308   ,   49.954525,    0.      ],
#       [   1.308   ,   50.003902,    0.      ],
#       [   1.308   ,   50.006992,    3.      ],
#       ...,
#       [ 845.946   , 1159.4888  ,   33.      ],
#      [ 845.946   , 1159.5037  ,    0.      ],
#       [ 845.946   , 1200.2019  ,    0.      ]], dtype=float32)
```

chambm, 2023-08-08:
I didn't notice before but the peakPicking filter syntax is slightly wrong. It mixes the old and new syntax which apparently just results in it silently doing nothing (not an ideal result!). You can either do:
peakPicking true 1
peakPicking vendor msLevel=1

This syntax change came when CWT peak picking was added and there needed to be a way to pick between local maxima peak picking and CWT (or vendor). CWT needed extra params, so key=value params were added for it. But experimenting with it now it's clear the syntax validation could be a lot better...

PierreSnell, 2023-08-08:
Yeah ! Putting `peakPicking vendor msLevel=1` indeed solves the problem for QC (continuum) files :D 

However, I still have the ion mobility 0 ones... However, for this one, I don't know if it's "normal" or not... 

As it's LC-IMS-MS, is it possible that the IMS is detecting something and not the MS ? 
We could say that this issue is closed and I will investigate with the chemists :) 

Thanks again for your help !

chambm, 2023-08-08:
You should get a warning in stderr that the vendor peak picking doesn't work for IMS. You can use CWT peak picking instead.

PierreSnell, 2023-08-08:
Indeed, sorry I was in INFO not DEBUG but I got the error message. 

However with `peakPicking cwt msLevel=1` (also tried capital CWT) or `peakPicking true 1` I still get the 
` [SpectrumList_Waters]: vendor centroiding is not supported for Waters ion mobility data`

Which raise the error but still convert the file and thus the data keep the 0...

chambm, 2023-08-08:
The 'true' version says you do want vendor centroiding, so it's expected to get the error there. You shouldn't get it for cwt though. I think that may be because you're putting peakPicking before lockmassRefiner. Lockmass refiner must either be the first filter or the second one after vendor peakPicking. Because lockmass refinement is a Waters API operation, we can't apply lockmass refinement to CWT peak picking, but we can apply CWT peak picking to lockmass refinement. This is indeed a confusing situation. It would be so nice if lockmass refinement could just be applied unconditionally by automatically knowing or finding the lockmass m/z.

PierreSnell, 2023-08-08:
Using peakPicking then lockmass I indeed got a `[SpectrumList_Waters]: vendor centroiding is not supported for Waters ion mobility data` 
Using lockmass then peakPicking I got  `MassLynxRawScanReader: Unable to open lockmass gain file.`
Using only peakPicking (no lockmass) I got no error and a good output (no more 0s) but if I understood your message correctly I would be able to do lockmass then peakPicking ? 

Thanks again for your help, it might be nice to add that to the [documentation](https://proteowizard.sourceforge.io/tools/msconvert.html) ? (as vendor cwt and possible combinations are not indicated).

chambm, 2023-08-09:
I think the "unable to open lockmass gain file" is probably an error related to the Docker setup or something, perhaps the .raw directory is mounted read-only? I don't get that error on Windows when I do lockmass followed by CWT on your HDMSe files. I'd assume that warning means the data isn't lockmass corrected.

I just made a PR to allow lockmass to work as the first filter, so peakPicking can come after it and you can choose vendor or cwt as the algorithm depending on whether it's IMS data or not.

chambm, 2023-08-09:
The Waters lockmass feature in the SDK creates/modifies the lmgt.inf file when lockmass correction is applied. So yes, for lockmass read-write is necessary for the .raw directory. That used to be the case for peak picking too but no longer.

closed 2023-08-14 by chambm