ProteoWizard/pwiz #2695: Peak picking for Ionic mobility data

PierreSnell, 2023-08-14:
Hello, sorry to post so many issues... 

When loading the converted Ionic mobility data, I realized that the Ionic Mobility column was dropped by the peak picking algorithm. 

Arguments : 
```
    arguments += f"wine msconvert {docker_file} "
    arguments += "--mzML "  # write mzML format [default]
    arguments += "--zlib "  # use zlib compression for binary data
    arguments += "--numpressAll "  # same as --numpressLinear --numpressSlof
    arguments += "--inten32 "  # encode intensity values in 32-bit precision
    arguments += "--mz32 "  # encode m/z values in 32-bit precision
    arguments += """--filter "msLevel 1" """
    arguments += """--filter "zeroSamples removeExtra" """

    #Option for Ionic mobility data (centroid, get peak picked by CWT)
        arguments += f"""--filter "lockmassRefiner mz={lock_mass} tol=0.5" """
        arguments += """--filter "peakPicking cwt msLevel=1" """
        arguments += "--combineIonMobilitySpectra "

    # Option for QC Files (centroid, get peak picked with vendor)
        arguments += """--filter "peakPicking vendor msLevel=1" """
        arguments += f"""--filter "lockmassRefiner mz={lock_mass} tol=0.5" """

    # Option for the rest of my datas
        arguments += f"""--filter "lockmassRefiner mz={lock_mass} tol=0.5" """
```

I then tried, for the ionic mobility argument, multiple possible combinations. 
With PyOpenMS: 

Lockmass & Peak picking & Combine mobility =>  3 columns, peak picked ~2 Millions data points (the IM column is all -1...)
Lockmass & Combine mobility => 4 columns, not peak picked, still continuous ~100 Millions data points
Peak picking & Combine mobility =>  3 columns, peak picked ~2 Millions data points (the IM column is all -1...)
Combine mobility =>  4 columns, not peak picked, still continuous ~100 Millions data points
Lockmass only => Error loading (Expected)
No filter => Error loading (Expected)

With Pyteomics: 
All error. 

Thanks in advance, 
Have a great day.

chambm, 2023-08-14:
Yes, CWT peak picking will drop the ion mobility array, it doesn't know what to do with that. If you need to keep it you just can't do peak picking. At least not with pwiz.

PierreSnell, 2023-08-14:
Without `combineIonMobilitySpectra` I got
```
Error writing run 1:
Invalid Scan Number
Conversion failed for 1 runs in MyFile.raw.
Error processing file MyFile.raw
```

So I couldn't test that

chambm, 2023-08-14:
Hmm not sure why non-combine mode is giving that error. Seems related only to the lockmass function (3) right? Try with filter scanEvent 1-2 to exclude those spectra.

PierreSnell, 2023-08-14:
I still got a `Invalid Scan Number` with `--filter scanEvent 1` or `--filter scanEvent 1-2`.

chambm, 2023-08-14:
As the first filter? Or second after lockmassRefiner?

chambm, 2023-08-14:
In the non-combined representation, there's only m/z and intensity arrays for each drift scan. The drift time is kept in a scan cvParam. So pyOpenMS would have to be coded to handle that. And I think they were one of the original ones to ask for the combined representation, so I'm not surprised they don't support the older one.

PierreSnell, 2023-08-14:
Thanks a lot for the info, I guess I will have to do a custom xml parser myself :) 

We can close this issue as I guess there is nothing you can do on your side (except coding the new best rt/mz/im peak detector haha).

Thanks again for the fast and good support, really appreciated !

closed 2023-08-14 by PierreSnell