ProteoWizard/pwiz #2761: Skyline: .wiff tests failing - is this expected?

meowcat, 2023-10-30:
I set up the Skyline environment as described in
https://skyline.ms/wiki/home/software/Skyline/page.view?name=HowToBuildSkylineTip

* I am using VS 2022 Community. 
* For now I don't have ReSharper but it should be irrelevant to the issue.
* I built all the stuff "around" Skyline with the b64.bat method
* In build64.log, I get the message: `[pwiz/data/vendor_readers/ABI/T2D] 64-bit MSVC build: Reader_ABI_T2D can only identify files, not read them.`

* I have selected "TestSettings_x64.runsettings".
* I have 916/936 tests passing. All the failing ones I checked have some error related to the Wiff reader:
```
Test method pwiz.SkylineTestData.PwizFileInfoTest.TestTicChromatogram threw exception: 
System.Exception: [WiffFile::create()] Could not load file or assembly 'Clearcore2.Data, Version=3.0.0.0, Culture=neutral, PublicKeyToken=2a79e0d8fd2e4eca' or one of its dependencies. The system cannot find the file specified.
```
* I have searched the pwiz issues as well as the Skyline support forum and not found the issue described.
* I have searched the Sciex webpage for anything obvious, and only found the "Sciex data converter" which contains Clearcore2.Data Version 2.something.
* Obviously I have the Clearcore dll's in my regular Skyline install, but I don't think the proper way is to just copy them over?

chambm, 2023-10-30:
The T2D message is expected. I suspect this has something to do with the fragile WIFF DLL load order problems we've seen when trying to switch to MSTest v2. The WIFF2 SDK is a SmartAssembly one with a single DLL that extracts embedded ones on the fly. Some of those embedded ones share names with the DLLs from the WIFF1 SDK. It's a miracle that it ever works to be honest. We have several times asked Sciex to give us a single SDK that can read both, or at least fix the assembly name overlap, but so far we're stuck with this situation.

What version of VS2022? I wonder if they switched MSTest to v2 in one of their updates.

chambm, 2023-10-30:
Interesting. I actually ran into that error now too. It only happens with a Debug config build of Skyline, not Release. Is that true for you as well?

meowcat, 2023-10-31:
Yes, true for me as well. Release seems to be working
The [instructions](https://skyline.ms/wiki/home/software/Skyline/page.view?name=HowToBuildSkylineTip) say:

> Be sure that your build configuration (Release x64) is set correctly. (Since VS 2022, Debug x64 is a better choice for debugging work.)

I find this a bit confusing; from this text I would have assumed that Debug x64 is fine to use, but does it mean that we need to code on Debug and test on Release?

meowcat, 2023-10-31:
VS Community 2022 64-bit 17.1.4

chambm, 2023-11-01:
Should be fixed in https://github.com/ProteoWizard/pwiz/pull/2764

closed 2024-01-18 by chambm