ProteoWizard/pwiz #2776: pyOpenMS does not read .mzML file

MathewGolding, 2023-11-14:
Hello, I recently installed ProteoWizard, thank you for maintaining this awesome tool for enabling open-source LCMS analysis, to use MSConvert (3.0.23318-6096a59 (automated build)), via the GUI, for converting .d (Agilent MassHunter) to the .mzML open file format. I am using the .mzML files to test and develop our automated LCMS package (MSCheck).

I am unable to open the files using pyOpenMS – I can confirm that the file does end with </indexedmzML>, have read #1310  to try resolve the issue myself along with reinstalling ProteoWizard. Have also confirmed the .d file is not corrupted with successful conversion (MSConvert, Version: 3.0.22290-bd9c20b (automated build)) and subsequent opening in pyOpenMS. Have tried to find the older version of ProteoWizard that worked (MSConvert ver=3.0.22290-bd9c20b) via Teamcity, but failed to find it. 

Any help massively appreciated!

MathewGolding, 2023-11-16:
[Attempt 6.zip](https://github.com/ProteoWizard/pwiz/files/13379038/Attempt.6.zip)
Find attached the different mzML conversion files from the prior mentioned older version and (as of writing) current version.
Take note that the modern version generates 15* as many spectra as the older version.

chambm, 2023-11-16:
The current version handles the EMR spectra (from the diode array detector) in your data where the old one doesn't. If that causes problems with pyOpenMS they should fix that, but you can also exclude them with a filter in msconvert. Use an msLevel filter where only levels 1- are accepted (EMR spectra are level 0).

MathewGolding, 2023-11-16:
Hi chambm, filtering this seems to have resolved the issue. Thank you very much for your help!

closed 2023-11-16 by MathewGolding