ProteoWizard/pwiz #2786: Failed to convert mzml files from .wiff, because of Illegal characters in path

FabRiols, 2023-11-24:
Hi guys,

I try using MS convert to convert a .wiff file to .mzml files, as I do all time.
However, this time something went wrong with Analyst software (Sciex acquisition soft.) when generating the very first acquisition, called blank, and makes MCconvert fail:

"Starting...
Opening file "D:\Analyst Data\Projects\20231122_P-22069_S-Perocchi_ALS_CSF_Plate04\Data\SLA Batch - 20231123 - 1.wiff" for read...
Failed - System.ArgumentException: error generating output filename for input file 'D:\Analyst Data\Projects\20231122_P-22069_S-Perocchi_ALS_CSF_Plate04\Data\SLA Batch - 20231123 - 1.wiff' and output run id 'SLA Batch - 20231123 - 1-Blank_10
Blank_10
Blank_10
Blank_10
blank_0000' ---> System.ArgumentException: Illegal characters in path."

This issue is specific to that sample. So my question, is it possible to skip this particular sample, so the software can manage the rest?
Thank you very much for your help!
Best wishes,
Fabien

brendanx67, 2023-11-24:
To me, that looks like there are newline characters in a sample name inside
your WIFF file. The file is:

Data\SLA Batch - 20231123 - 1.wiff

And the sample name is:

"Blank_10
Blank_10
Blank_10
Blank_10
blank_0000"

And MSConvert is trying to create a separate mzML file for each sample in
your WIFF file by using the naming strategy:

<wiff-basename>-<sample-name>.mzML

I am not sure what this code currently does to remove invalid path
characters from the <sample-name> but it seems that code needs to be
improved to include \n and \r and any other invalid path characters it may
have omitted.

I am sure this will be a relatively straightforward fix that can be
released soon. In the meantime, if SCIEX offers any ability to edit sample
names in a WIFF file, you could fix this on your side. And probably you
should look into how you created such an odd and undesirable sample name in
the first place and avoid doing it again.

Thanks for posting your issue so clearly.

chambm, 2023-11-30:
Thanks for the report. I forgot to update MSConvertGUI's logic when I did msconvert's.

closed 2023-11-30 by chambm