ProteoWizard/pwiz #2802: Latest installation instructions

tejalal, 2023-12-04:
I was trying to experiment with the ProteoWizard tools (specifically Skyline and MSConverGUI). I didn't find the installation instructions that one can follow to setup the required environment for the git repo. I want to do the setup in the local system.
I came to know across the Skyline installation instruction given here [https://skyline.ms/wiki/home/software/Skyline/page.view?name=HowToBuildSkylineTip](url).

I was able to run the Skyline from the solution file, but the execution of the MSConvert from the solution file failed. It failed with error like CLI could not be imported and many other errors.

I also came across ProteoWizard developer page at [https://proteowizard.sourceforge.io/doc_developers.html](url) but it also has old instructions. 

Are there different instruction to build the pwiz and MSConvert solution file.
Thanks!

nickshulman, 2023-12-05:
When I want to build MSConvert and SeeMS, the target that I tell quickbuild.bat to build is "executables".

The commandline I use is:
`quickbuild.bat --i-agree-to-the-vendor-licenses -j2 toolset=msvc-14.3 address-model=64 --without-compassxtract executables`

My machine is configured to be able to build Skyline, and there was no additional setup required to build MSConvert and SeeMS.

If you are still having trouble building you should attach your build output and we may be able to figure out what is going wrong.

tejalal, 2023-12-08:
Thanks @nickshulman and @chambm, 
Following your suggestions, I did a fresh installation, cloned the repo, and ran the above command.
Build was still failing due to CustomDataSourceDialog import error, so after investigation I changed the pwiz_tools/MSConvertGUI/CustomDataSourceDialog/CustomDataSourceDialog.csproj file from 
`<TargetFrameworkVersion>v4.0</TargetFrameworkVersion>` to
`<TargetFrameworkVersion>4.7.2</TargetFrameworkVersion>` and it worked in the release mode.

I want to execute the project in the bebug mode. Actually, I want to capture the input of the OverlapMultiplexer that is defined in the Skyline. Also, when I import Skyline in MSConvert then it doesn't works. Any suggestions on it would be helpful.

closed 2023-12-08 by tejalal

nickshulman, 2023-12-08:
We recommend that you avoid the class "OverlapDemultiplexer" in Skyline.
That class was written before we understood the proper way to demultiplex overlapping isolation windows.
The demultiplexing that MSConvert does is much better than the thing that Skyline knows how to do.

That is, in the "Edit Isolation Scheme" dialog in Skyline, you should never choose anything other than "None" in the "Deconvolution" dropdown.
Instead, you should use MSConvert to generate demultiplexed .mzML files, and Skyline should not be told that the isolation windows ever overlapped.

tejalal, 2023-12-11:
OK, thanks @nickshulman it was really helpful. One more thing, is there any way to plot the original multiplexed input and demultiplexed output generated by MSConvert. I checked SeeMS, but didn't found any option for it.
Thanks!

chambm, 2023-12-14:
The demultiplexing generates a different number of spectra, so it's tricky to directly compare them. But SeeMS will let you do it with a bit of work. The right click menu in SeeMS' spectrum list lets you overlay or stack spectrum graphs (on the current graph or a new tab). So if you view a spectrum in the original file, then go to the demultiplexed file and find the same spectrum (it will have the same "scan=XXX" number, there will be two versions (demux=0 and demux=1 IIRC). Select both demux=0 and demux=1 rows and right click, then select "Overlap on current graph" or "Stack on current graph".