ProteoWizard/pwiz #2864: Thermo RawFileReader library is out-of-date

FarmGeek4Life, 2024-02-08:
Currently used version of the Thermo RawFileReader library is 5.0.6, from 2019, and there is no direct support for the Astral instrument in that version.

Latest versions are 5.0.0.93 (for .NET Framework 4.7.1 and NETStandard 2.0) or 7.1.21 (for .NET 5.0).

5.0.0.93 does include a mass analyzer type `ASTMS` for the scan filters.

alchemistmatt, 2024-02-08:
Converting an Astral .raw file to .mzML leads to this warning message:
```
no matching instrument configuration for analyzer type Unknown
```
If you update to the newer Thermo RawFileReader library, and update the backend source code, you should be able to determine the analyzer type. 

Example scan filter:
`ASTMS + c NSI d Full ms2 592.6334@hcd21.42 [150.0000-1500.0000]`

If you need an example .raw file, I can provide one, but it's large (8 GB)

chambm, 2024-02-08:
You're getting unknown analyzer with the latest pwiz version? I know about the newer RFR library and its support for ASTMS in the filter line, but I thought the work around I have in there now would suffice for the time being.

alchemistmatt, 2024-02-08:
Yes, with the latest build from http://teamcity.labkey.org/viewType.html?buildTypeId=bt83

```
C:\DMS_Programs\ProteoWizard\msconvert.exe
ProteoWizard release: 3.0.24038 (85b7ba5)
Build date: Feb  7 2024 22:39:35

C:\DMS_Programs\ProteoWizard\msconvert.exe 20221216_OLEP05_HAP1_500ng_DDA_120k_10ms_1sec_5ms_v6_30min_1.raw
format: mzML
    m/z: Compression-None, 64-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: .
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:

chromatogram list filters:

filenames:
  20221216_OLEP05_HAP1_500ng_DDA_120k_10ms_1sec_5ms_v6_30min_1.raw

processing file: 20221216_OLEP05_HAP1_500ng_DDA_120k_10ms_1sec_5ms_v6_30min_1.raw
calculating source file checksums

writing output file: .\20221216_OLEP05_HAP1_500ng_DDA_120k_10ms_1sec_5ms_v6_30min_1.mzML
no matching instrument configuration for analyzer type Unknown
```

Here's an excerpt from the start of the .mzML file
```
    <fileDescription>
      <fileContent>
        <cvParam cvRef="MS" accession="MS:1000579" name="MS1 spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
      </fileContent>
      <sourceFileList count="1">
        <sourceFile id="RAW1" name="20221216_OLEP05_HAP1_500ng_DDA_120k_10ms_1sec_5ms_v6_30min_1.raw" location="file:///">
          <cvParam cvRef="MS" accession="MS:1000768" name="Thermo nativeID format" value=""/>
          <cvParam cvRef="MS" accession="MS:1000563" name="Thermo RAW format" value=""/>
          <cvParam cvRef="MS" accession="MS:1000569" name="SHA-1" value="f2a14587c2b4ad9417734f66a9b73e6d41374699"/>
        </sourceFile>
      </sourceFileList>
    </fileDescription>
    <referenceableParamGroupList count="1">
      <referenceableParamGroup id="CommonInstrumentParams">
        <cvParam cvRef="MS" accession="MS:1000492" name="Thermo Electron instrument model" value=""/>
        <cvParam cvRef="MS" accession="MS:1000529" name="instrument serial number" value="OLEP05"/>
        <userParam name="instrument model" value="Olympus-based mass spectrometer"/>
      </referenceableParamGroup>
    </referenceableParamGroupList>
    <sampleList count="1">
      <sample id="R_x003a_A1" name="">
        <cvParam cvRef="MS" accession="MS:1000002" name="sample name" value="R:A1"/>
      </sample>
    </sampleList>
    <softwareList count="2">
      <software id="Xcalibur" version="0.7-0.7.77.0-flavor=debug/0.7.39.0">
        <cvParam cvRef="MS" accession="MS:1000532" name="Xcalibur" value=""/>
      </software>
      <software id="pwiz" version="3.0.24038">
        <cvParam cvRef="MS" accession="MS:1000615" name="ProteoWizard software" value=""/>
      </software>
    </softwareList>
    <instrumentConfigurationList count="1">
      <instrumentConfiguration id="IC1">
        <referenceableParamGroupRef ref="CommonInstrumentParams"/>
        <componentList count="3">
          <source order="1">
            <cvParam cvRef="MS" accession="MS:1000398" name="nanoelectrospray" value=""/>
            <cvParam cvRef="MS" accession="MS:1000485" name="nanospray inlet" value=""/>
          </source>
          <analyzer order="2">
            <cvParam cvRef="MS" accession="MS:1000079" name="fourier transform ion cyclotron resonance mass spectrometer" value=""/>
          </analyzer>
          <detector order="3">
            <cvParam cvRef="MS" accession="MS:1000624" name="inductive detector" value=""/>
          </detector>
        </componentList>
        <softwareRef ref="Xcalibur"/>
      </instrumentConfiguration>
    </instrumentConfigurationList>
    <dataProcessingList count="1">
      <dataProcessing id="pwiz_Reader_Thermo_conversion">
        <processingMethod order="0" softwareRef="pwiz">
          <cvParam cvRef="MS" accession="MS:1000544" name="Conversion to mzML" value=""/>
        </processingMethod>
      </dataProcessing>
    </dataProcessingList>
```

chambm, 2024-02-08:
Is that a production Astral and/or production firmware? The "Olympus-based mass spectrometer" makes me think not. That's why my workaround isn't working. Time to update the library it seems.

closed 2024-02-09 by chambm