ProteoWizard/pwiz #2895: Fails to build on Linux with missing librt

alansill, 2024-03-11:
Starting from a fresh git clone of the repository on Centos 8 Linux using either the default gcc 8.3.1 or a newer gcc 10.2.0, I encounter the same error trying to build anything:

```gcc.link /home/(my userid)/pwiz/build-linux-x86_64/ext/hdf5/gcc-10/release/link-static/runtime-link-static/threading-multi/H5make_libsettings
/usr/bin/ld: cannot find -lrt
collect2: error: ld returned 1 exit status

    "g++"    -o "/home/(my userid)/pwiz/build-linux-x86_64/ext/hdf5/gcc-10/release/link-static/runtime-link-static/threading-multi/H5make_libsettings" -Wl,--start-group "/home/asill/pwiz/build-linux-x86_64/ext/hdf5/gcc-10/release/link-static/threading-multi/H5make_libsettings.o"     -lrt -Wl,--end-group -pthread -static -fPIC -fno-pie -no-pie

...failed gcc.link /home/(my userid)/pwiz/build-linux-x86_64/ext/hdf5/gcc-10/release/link-static/runtime-link-static/threading-multi/H5make_libsettings...
```

Note librt is available on our system. Is this a version problem? How can I work around this?

```$ ls -l /usr/lib*/librt*
-rwxr-xr-x 1 root root 142608 Feb  4  2020 /usr/lib64/librt-2.28.so
lrwxrwxrwx 1 root root     22 Feb  4  2020 /usr/lib64/librt.so -> ../../lib64/librt.so.1
lrwxrwxrwx 1 root root     13 Feb  4  2020 /usr/lib64/librt.so.1 -> librt-2.28.so
```

chambm, 2024-03-11:
It may want to link statically, so maybe `yum install glibc-static` ?

alansill, 2024-03-19:
After exploring several alternatives, we decided to add glibc-static to the login and worker nodes of our cluster as the simplest alternative. Note that static linking is often discouraged these days, see https://best.openssf.org/Compiler-Hardening-Guides/Compiler-Options-Hardening-Guide-for-C-and-C++.html#when-not-to-use-5 but it seems the best approach in this case. Thanks for the suggestion.

closed 2024-03-19 by alansill