ProteoWizard/pwiz #2897: Pwiz build failed using GCC 13 on Linux

nekoteoj, 2024-03-11:
I tried to build the latest commit of pwiz using gcc 13 and encountered some errors. I guessed that there are some part in the current source code that are interfering with the c++ libraries included in gcc 13.

Here is the procedure to reproduce the error message. (I tried to reproduce what is in the repository's github workflow file)
```sh
git clone https://github.com/ProteoWizard/pwiz.git
cd pwiz
./quickbuild.sh --abbreviate-paths --i-agree-to-the-vendor-licenses address-model=64 pwiz_tools/BiblioSpec toolset=gcc
```

Here is the error message
```
gcc.compile.c++ /home/pisit/Documents/pwiz/build-linux-x86_64/pwiz/data/common/gcc-13/rls/adrs-mdl-64/lnk-sttc/thrd-mlt/diff_std.o
In file included from /home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/tools/mp.hpp:341,
                 from /home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/policies/policy.hpp:11,
                 from /home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/special_functions/math_fwd.hpp:31,
                 from /home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/special_functions/modf.hpp:13,
                 from ./pwiz/utility/math/round.hpp:30,
                 from /home/pisit/Documents/pwiz/pwiz/data/common/diff_std.hpp:28,
                 from /home/pisit/Documents/pwiz/pwiz/data/common/diff_std.cpp:26:
/usr/include/c++/13.2/utility:98:14: error: '__and_' was not declared in this scope; did you mean 'std::__and_'?
   98 |     noexcept(__and_<is_nothrow_move_constructible<_Tp>,
      |              ^~~~~~
      |              std::__and_
In file included from /usr/include/c++/13.2/bits/move.h:37,
                 from /usr/include/c++/13.2/bits/exception_ptr.h:41,
                 from /usr/include/c++/13.2/exception:164,
                 from /usr/include/c++/13.2/stdexcept:38,
                 from ./pwiz/utility/math/round.hpp:29:
/usr/include/c++/13.2/type_traits:176:12: note: 'std::__and_' declared here
  176 |     struct __and_
      |            ^~~~~~
/usr/include/c++/13.2/utility:98:21: error: 'is_nothrow_move_constructible' was not declared in this scope; did you mean 'std::is_nothrow_move_constructible'?
   98 |     noexcept(__and_<is_nothrow_move_constructible<_Tp>,
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     std::is_nothrow_move_constructible
/usr/include/c++/13.2/type_traits:1140:12: note: 'std::is_nothrow_move_constructible' declared here
 1140 |     struct is_nothrow_move_constructible
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13.2/utility:98:54: error: expected primary-expression before '>' token
   98 |     noexcept(__and_<is_nothrow_move_constructible<_Tp>,
      |                                                      ^
/usr/include/c++/13.2/utility:98:55: error: expected primary-expression before ',' token
   98 |     noexcept(__and_<is_nothrow_move_constructible<_Tp>,
      |                                                       ^
/usr/include/c++/13.2/utility:98:55: error: expected ')' before ',' token
   98 |     noexcept(__and_<is_nothrow_move_constructible<_Tp>,
      |             ~                                         ^
      |                                                       )
/usr/include/c++/13.2/utility:98:55: error: expected ';' before ',' token
   98 |     noexcept(__and_<is_nothrow_move_constructible<_Tp>,
      |                                                       ^
      |                                                       ;
/home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/tools/mp.hpp:342:15: error: 'boost::math::tools::meta_programming::std::size_t' has not been declared
  342 | template<std::size_t... I>
      |               ^~~~~~
/home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/tools/mp.hpp:343:29: error: 'index_sequence' in namespace 'boost::math::tools::meta_programming::std' does not name a template type
  343 | using index_sequence = std::index_sequence<I...>;
      |                             ^~~~~~~~~~~~~~
/home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/tools/mp.hpp:345:15: error: 'boost::math::tools::meta_programming::std::size_t' has not been declared
  345 | template<std::size_t N>
      |               ^~~~~~
/home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/tools/mp.hpp:346:34: error: 'make_index_sequence' in namespace 'boost::math::tools::meta_programming::std' does not name a template type
  346 | using make_index_sequence = std::make_index_sequence<N>;
      |                                  ^~~~~~~~~~~~~~~~~~~
/home/pisit/Documents/pwiz/libraries/boost_1_76_0/boost/math/tools/mp.hpp:349:33: error: 'index_sequence_for' in namespace 'boost::math::tools::meta_programming::std' does not name a template type
  349 | using index_sequence_for = std::index_sequence_for<T...>;
      |                                 ^~~~~~~~~~~~~~~~~~

```

I'm using Void Linux (x86-64) with GCC 13.2.0 but the problem existed for other distros as well. I think it is the compiler things. Here is the compiler information.
```
g++ (GCC) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The code built fine with gcc 11. I think it would be great to make this great library to support the newer version of gcc too. Thanks!

CharlesEGrant, 2024-11-12:
For the same reason, I haven't been able to build ProteoWizard with any GCC newer than major version 11.

The issue seems to be with incompatibilities between BOOST 1.76 and newer compiler versions. Among other issues boost/math/tools/mp.h tries to #include <utility> inside of a namespace, which end up [violating the C++ standard](https://eel.is/c++draft/requirements#using.headers-3). Older compilers overlooked this, but it's been in the C++ standard for a while, and newer compilers are enforcing it. [Boost fixes this in 1.77](https://github.com/boostorg/math/commit/1ec5c98d80de97f9e962c5627e1a0e6096099894).

This is becoming more of a problem as more distributions are starting to default to GCC versions > 12. Any chance of upgrading ProteoWizard to at least 1.77?

chambm, 2025-01-15:
Upgraded to Boost 1.86 yesterday!