ProteoWizard/pwiz #2934: High intensity TIC SeeMS compared to other softwares

viictoriaeriksson, 2024-04-19:
Have used MSconvert (with peak picking) to export my data measured on VION-IMS-QTOF (coupled with LC) out from UNIFI.
The converted file (.mzML) have there after been opened in different softwares to show the TIC. When looking at the TIC in MS-DIAL (after processing the file), ToppView (from OpenMS), and a homemade R-script, the TIC look identical and have the same intensities across the whole measurement. However, when I open the file in SeeMS, I notice the the "shape" of the TIC looks similar to the other softwares, however, the intensities are higher. The other have a max intensity 1.7*10^6 for the TIC, but in SeeMS the max intensity is over 2.5*10^6.
I have compared the individual intensities of the ions for some of the scans, and they are the same between SeeMS and the others. Hence, the difference in the TIC, must come from how the TIC is created. 

Therefore, I was wondering what could be the reason for the higher TIC intensities in SeeMS? Does it use some special method to construct it, other than just sum the intensities of the ions in all the scans for one retention time?
(see attached pictures of one data file's TIC in MSDIAL and SeeMS)
[image]
[image]

nickshulman, 2024-04-19:
Can you send us your mzML file?
If the file is too large to attach to this Issue you can upload it here:
https://skyline.ms/files.url

viictoriaeriksson, 2024-04-19:
I have now uploaded my mzML-file to skyline.ms/files.url, named "Drugstd-200ppb_1_A,3_1.zip".

nickshulman, 2024-04-19:
The mzML file contains a TIC chromatogram which is what you see in SeeMS.exe.
The TIC chromatogram is basically a list of retention times and intensities.
The maximum intensity in the TIC chromatogram is 2.981x10^6 at retention time 4.635 minutes.

The other number that you are seeing, 1.775x10^6 is what you would get if you were to add up all of the intensities of all of the spectra with a particular retention time (the number might be a little different depending on the way that floating point numbers get rounded).

I do not know exactly why those numbers are different. It might be that the intensities in the spectra changed a little when the data in the spectra was centroided.

When you open the mzML file in SeeMS.exe you are able to see the TIC chromatogram immediately because SeeMS just reads the chromatogram data from a spot near the end of the file.
I suspect that the other software that you are using takes a long time to show you their TIC chromatogram because they have to read each of the spectra and add up all of the intensities.

Does this answer your questions?

brendanx67, 2024-04-19:
When you say "with peak picking" is that using the vendor algorithm and does the file also include profile spectra? From Nick's response, I would expect the TIC to look the same if you simply opened the original raw data file. Can any of the other programs you listed do that? It will be good to get Matt to weigh in here, but I think the TIC chromatogram is coming from Waters and would likely appear similar to what you might see in Waters native software.

chambm, 2024-04-22:
As Nick said, the SeeMS chromatogram is coming from the vendor API or the file. It's never generated on the fly. And it's usually generated from spectrum-by-spectrum metadata for TIC, which is usually different than the sum of the centroided peak intensities (like you said, it's the sum of the profile intensities).

closed 2026-01-14 by chambm