ProteoWizard/pwiz #2991: ProteoWizard build issues under Windows 11

keesh0, 2024-05-29:
Hello,
I am trying to evaluate adding new per scan metrics to the Quameter executable.
Specifically, the following:
[image]

However, I cannot build the master branch from source under Windows 11 using:
quickbuild.bat address-model=64  --i-agree-to-the-vendor-licenses  
or
b2.exe
or
Visual Studio 2015, 2019, or 2022 pwiz.sln solutions.
Has anybody successfully built quameter.exe with full debug symbols under Windows 11?
Most C++ projects that I have worked on have provided a convenient CMake build system to ease building cross platform.

chambm, 2024-05-29:
It seems to be defaulting to VS 2015. Try adding `toolset=msvc-14.3` to the command.

keesh0, 2024-05-30:
Things are better with:
>quickbuild.bat address-model=64 --i-agree-to-the-vendor-licenses toolset=msvc-14.3 -q  --debug-building pwiz_tools\Bumbershoot\quameter//install

But,
...skipped <pD:\source\keesh_programming\mayo2\Lab\projects\proteo\ProteoWizard\build-nt-x86\msvc-release-x86_64>quameter.exe for lack of <pD:\source\keesh_programming\mayo2\Lab\projects\proteo\ProteoWizard\build-nt-x86\pwiz_tools\Bumbershoot\quameter\msvc-14.3\release\address-model-64\asynch-exceptions-on\link-static\threading-multi>quameter.exe...

Also, how do I generate a full symbol debug build?

chambm, 2024-05-30:
If there's a ...skipped there should be a ...failed target somewhere in the output indicating what dependency failed to build. --debug-building is for debugging the actual build system (Boost Build Jamfiles), not building debug targets. For a full debug build, add `debug` to the command. For a release build with symbols, add `debug-symbols=on` (without adding `debug`). For a fast debug build, add `debug runtime-debugging=off iterator-debugging=off`.

keesh0, 2024-05-30:
You are correct:
projects\proteo\ProteoWizard\build-nt-x86\ext\hdf5\msvc-14.3\release\address-model-64\asynch-exceptions-on\link-static\threading-multi\H5detect.obj : fatal error LNK1112: module machine type 'x86' conflicts with target machine type 'x64'

keesh0, 2024-05-30:
Building and running are looking positive.

Debugging...
Since I don't have a Visual Studio project for Quameter,
generating a PDB file would be nice (if that is an option)?

chambm, 2024-05-30:
To debug under Visual Studio, run (from a fresh cmd prompt):
`libraries\msvc_setup_script.bat x64 143`
`devenv /debugexe build-nt-x86\msvc-debug-x86_64\quameter.exe <program arguments>`
Once you've got VS open, you can change the arguments as if they were "project settings" (solution explorer). F5 and Ctrl-F5 work as normal.

If you build with debug or debug-symbols=on then you'll have PDB files. Visual Studio will pick them up automatically when you run under the debugger.

keesh0, 2024-05-31:
D:\source\keesh_programming\mayo2\Lab\projects\proteo\ProteoWizard>libraries\msvc_setup_script.bat x64 1
43 devenv /debugexe build-nt-x86\msvc-debug-x86_64\quameter.exe
Found with vswhere C:\Program Files\Microsoft Visual Studio\2022\Community
The input line is too long.
The syntax of the command is incorrect

chambm, 2024-05-31:
The batch file call and the devenv call must be two separate lines.

chambm, 2024-05-31:
Weird, not sure what's going on there. That script is just a more flexible version of "Developer Command Prompt for VS 2022" (because it works for 3 different VS versions and both x86 or x64). You can find that in your start menu. Use that shortcut to open a cmd prompt and then do the second command `devenv`.

chambm, 2024-05-31:
(after changing directory to the pwiz repo of course)

keesh0, 2024-05-31:
It would have been much easier to provide CMake files which could auto generate solutions / projects. 
VS 2022 launches with no source files.
When I open the main quameter file, I cannot launch the debugger.

What is the following supposed to do as my BAT skills are rusty?\
libraries\msvc_setup_script.bat x64 143

keesh0, 2024-05-31:
Made it further, but cannot step for some reason?
[image]

chambm, 2024-05-31:
That command will bring the Visual Studio 2022 (VC++ 14.3) tools into your PATH:
> That script is just a more flexible version of "Developer Command Prompt for VS 2022" (because it works for 3 different VS versions and both x86 or x64).

I am not a fan of CMake FYI.

> Made it further, but cannot step for some reason?
What did you do to start debugging?

keesh0, 2024-05-31:
I following your instructions to start debugging.
I can see all variable values, but cannot step.
[image]

Note, the PDB and EXE are in a separate folder than the source code:
D:\source\keesh_programming\mayo2\Lab\projects\proteo\ProteoWizard\build-nt-x86\pwiz_tools\Bumbershoot\quameter\msvc-14.3\debug\address-model-64\asynch-exceptions-on\link-static\threading-multi

D:\source\keesh_programming\mayo2\Lab\projects\proteo\ProteoWizard\pwiz_tools\Bumbershoot\quameter

chambm, 2024-05-31:
Looks like it's been fixed in a Visual Studio update:
https://developercommunity.visualstudio.com/t/debug-single-step-function-does-not-work/10521835

keesh0, 2024-05-31:
Thanks for going the extra mile.
Hopefully I can contribute something useful to this project.

keesh0, 2024-06-12:
I was able to implement this feature by making only minor changes to quameter.h, .cpp in my copy of the master branch.
I am testing on additional input data.
Hopefully, I can issue a pull request soon.

Here is the motivation.
Peak shapes are expected to be Gaussian in shape thereby having a symmetric shape around the apex. While evaluating new columns and solvents it was observed that several columns provided narrow peaks with nearly vertical ascending peak shape.
The asymmetrical peak shape did not allow the mass spectrometer enough time to trigger on front side of peaks. Once the apex occurred then the instrument was able to trigger. This alters the ability to assess peak area and contributes to more variability in the assays.
This asymmetry artifact was variable as a few broader peaks indicated traditionally gaussian peak shapes and received disproportionately more MS2 triggers which in turn inflates the spectral counts for that protein. Many proteins like the ApoE protein above were negatively impacted throughout the gradient while other proteins were inflated. This in-equality affects our ability to assume that more spectral counts means a more abundant protein.

Example file (de-identified)
[image]

chambm, 2024-06-21:
Looking forward to a PR.

closed 2024-06-21 by chambm

keesh0, 2024-06-21:
Thanks.
I added a small amount of code after the second CrawdadPeakFinder call in NISTMSMetrics() thinking that is what we want.
But we are not seeing the asymmetry expected using FWHM.
They originally wanted the asymmetry ratio calculated at 0.1 of max peak.
But prior to trying the above more invasive change, 
I was wondering the differences of the two (2) CrawdadPeakFinder calls in NISTMSMetrics().
What is the first call used for?

thx e.-

`    // Metric C-3A: Median peak width of identified spectra
    // Metric C-3B: IQR of peak width of identified spectra
    {
        accs::accumulator_set<double, accs::stats<accs::tag::percentile> > identifiedPeakWidths;
        BOOST_FOREACH(const XICWindow& distinctMatch, pepWindow)
            if (distinctMatch.bestPeak)
                // BEG KEESH MAYO ASYMM METRIC
                for (size_t i = 0; i < distinctMatch.PSMs.size(); ++i) // the best peak is shared between all PSMs
                {  
                    identifiedPeakWidths(distinctMatch.bestPeak->fwhm);
                    // TODO-- Make sure that we have the right peaks 
                    double startTime = distinctMatch.bestPeak->startTime / 60.0;  // convert seconds to minutes
                    double peakTime = distinctMatch.bestPeak->peakTime / 60.0;
                    double endTime = distinctMatch.bestPeak->endTime / 60.0;
                    double a = peakTime - startTime;
                    double b = endTime - peakTime;
                    double epsilon = boost::math::tools::epsilon<double>();
                    // Perform the division safely
                    double As = b / (a + epsilon);
                    surviving_peaks[peakTime] = As;
                }
                // END KEESH MAYO ASYMM METRIC
        identifiedPeakWidthMedian = accs::percentile(identifiedPeakWidths, accs::percentile_number = 50);
        identifiedPeakWidthIQR = accs::percentile(identifiedPeakWidths, accs::percentile_number = 75) -
                                 accs::percentile(identifiedPeakWidths, accs::percentile_number = 25);
    }`

chambm, 2024-06-21:
The first one is finding peaks for chromatograms of identified peptides. The second one is finding peaks for chromatograms of unidentified spectra, grouped by precursor m/z IIRC.

brendanx67, 2024-06-21:
I highly recommend you have a look at Panorama AutoQC for instrument system
suitability track and sample QC.

https://pubs.acs.org/doi/full/10.1021/acs.jproteome.6b00744

https://panoramaweb.org/home/wiki-page.view?name=qc_with_panorama

This is actively supported and funded software. While the NIST metrics are
not, and not something we can put very much time into supporting.

It is what the MacCoss lab itself and multiple pharma companies are
actively using. We have recently implemented Kurtosis and peak Standard Dev
for asymmetry and width tracking.

keesh0, 2024-08-02:
Still waiting on the lab to make an exact decision on what to output for plotting...

keesh0, 2024-08-21:
The lab has sanctioned our changes.  I will issue a PR when we undergo testing.
What branch should I issue the PR to?

chambm, 2024-08-21:
Fork the repo and make a PR to master. The branch in your repo doesn't matter AFAIK.