ProteoWizard/pwiz #2994: Extraction of collision energy for Sciex 7600 scans in mzML format

chrashwood, 2024-05-31:
Dear Proteowizard team,

Hope you're all doing well. I've grown accustomed to having collision energy encoded in the mzML files of Thermo raw files however there does not appear to be an equivalent for the Sciex 7600. Is this a feasible request?

When I review the data in Analyst (Sciex proprietary software for reading their raw files), the table of the m/z targets and collision energies is provided, so it would appear that they are encoded in the raw files at the very least.

In regards to benefits of having collision energy stated in the mzML files, Skyline has a spectrum filter that allows users to separate chromatograms by their collision energy. This is a very useful feature when performing collision energy optimisation, e.g. 10 different collision energies on the same precursor target within a single run.

Cheers,
Chris

chambm, 2024-06-06:
ProteoWizard can't get CE from WIFF1 files due to API limitations. But it should be able to get it from WIFF2 files. The WIFF1 API has CE properties in it but for some reason they seem to always be 0 for MS2 scans. I'm not sure that's the same as what you're referring to with "the table of the m/z targets and collision energies" - that sounds more like looking at the instrument method, which it doesn't seem the API allows either. But if you want to share an example WIFF1 file I can double check that the CE is inaccessible.

chrashwood, 2024-06-06:
Thanks for looking into this. It's structured as an "MRM HR" (PRM) experiment where we have multiple collision energies for a few precursors, as we are doing collision energy optimization. The table I'm mentioning is the method information that can be read from Analyst when accessing the specific .wiff files (I think it opens all of them).

I've uploaded the all the acquisition files to the Skyline share portal (https://skyline.ms/project/home/support/file%20sharing/begin.view) and they're in the folder "20240607_ForMattChambers_SCIEXCEExtraction".

Thank you for taking a look at this.

chrashwood, 2024-06-27:
Resolved, I can now see the collision energy now in the .wiff1 files. Not sure if it's from a Proteowizard update but it's working now.
```

              <activation>
                <cvParam cvRef="MS" accession="MS:1000422" name="beam-type collision-induced dissociation" value=""/>
                <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="80.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
              </activation>
```

chambm, 2024-06-27:
Right, I fixed this in #3023 .