ProteoWizard/pwiz #3012: Please fix Agilent .d derived mzML spectrum scan numbers to be sequential starting at 1

Massmz, 2024-06-13:
This seems like it has been an issue for many years. The Agilent scan id numbers are large non-sequential numbers so using the MSConvert scan number filter doesn't work. The OpenMS FileConverter utility produces an mzML file with sequential scans starting at 1 but does not have a scan number filter. Can this finally be remedied in MSConvert?

chambm, 2024-06-13:
That's the nativeID. It's an integer used to get a specific spectrum in the Agilent API. There was some API limitation that made it a good reason for using that non-sequential integer instead of a sequential one. But I can't remember the exact reason/limitation. The thing to fix is here is the lack of index subset filter in msconvert GUI. The command-line has it though. However the index is filter is 0-based.

Massmz, 2024-06-13:
Thanks. BTW, I am going from MassHunter generated mzData files to mzML with the FileConverter tool not the raw .d file folders.