ProteoWizard/pwiz #3127: Error converting .wiff2 format to .mzML format

renkedawei, 2024-08-19:
I used the SCIEX X500R instrument to obtain the raw data in.wiff2 format, but when I used msconvert to convert it, it reported an error. Here are the details of the error:
[Reader_ABI::fillInMetadata] unable to determine instrument model ([WiffFile2Impl::getInstrumentModel()] æªå°å¯¹è±¡å¼ç¨è®¾ç½®å°å¯¹è±¡çå®ä¾ã)
[Reader_ABI::fillInMetadata] unable to read instrument serial number ([WiffFile2Impl::getInstrumentSerialNumber()] æªå°å¯¹è±¡å¼ç¨è®¾ç½®å°å¯¹è±¡çå®ä¾ã)
[Reader_ABI::fillInMetadata] unable to read sample acquisition time ([WiffFile2Impl::getSampleAcquisitionTime()] String å¼ç¨æ²¡æè®¾ç½®ä¸º String çå®ä¾ã
Calculating SHA1 checksum...
Processing...
Failed - System.Exception: [Spectrum2Impl::getDataSize()] null spectrum
   在 pwiz.CLI.msdata.SpectrumList.empty()
   在 MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   在 MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

chambm, 2024-08-19:
It's hard to imagine this isn't a corrupt file, or at least a corrupt sample. Can you share the file?

renkedawei, 2024-08-20:
> 很难想象这不是一个损坏的文件，或者至少是一个损坏的样本。你能分享这个文件吗？

I have sent the file to your email address (matt.chambers42@gmail.com), thank you for your timely reply

renkedawei, 2024-08-20:
> It's hard to imagine this isn't a corrupt file, or at least a corrupt sample. Can you share the file?

I have sent the file to your email address (matt.chambers42@gmail.com), thank you for your timely reply!