ProteoWizard/pwiz #3159: msconvert not extracting full name of scans for first and last entries of .wiff file in v3.0.2

ChristianIeritano, 2024-09-15:
Hi proteowizard friends, 

We use msconvert to convert scans saved to .wiff files (from a SCIEX instrument) to .mzML. The following script is part of a [GUI](https://github.com/WaterFEL/SpectroGUI/blob/master/GUI/workflows/wiff2mzml.py) we use to process UVPD data measured on a modified SCIEX 5500, which works fine w/ the msconvert.exe found in ProteoWizard 3.0.11392, but does not work for any v3.0.2 variant. Specifically, when processing a .wiff and associated .wiff.scan file, the full filename of the first and last scans is truncated. All other scans are extracted and named normally. 

Note that the filenames contain some verbose info about the filenames and a -### before the .wiff(.scan) extension. An example .wiff file for testing can be found at [our GitHub](https://github.com/WaterFEL/SpectroGUI/tree/master/Example). 

Thanks!

```python
import os, traceback, subprocess, time
from PyQt6.QtWidgets import QApplication

def convert_wiff_to_mzml(wiff_file, directory, mzml_directory):
    ''' Function to convert .wiff files to .mzml using msconvert
    input is .wiff file, directory that contains .wiff files, and directory to output mzml files to'''
    
    #check if required files are present
    wiff_file_check = os.path.join(directory, wiff_file)
    scan_file_check = f'{os.path.join(directory, wiff_file)}.scan'
    
    if not os.path.exists(wiff_file_check):
        print(f'The corresponding .wiff file is missing from the directory. Please add the following file to the directory, and re-run the code:\n{os.path.basename(wiff_file_check)}\n\n')
        QApplication.processEvents()
        return False

    if not os.path.exists(scan_file_check):
        print(f'The corresponding .scan file is missing from the directory. Please add the following file to the directory, and re-run the code:\n{os.path.basename(scan_file_check)}\n\n')
        QApplication.processEvents()
        return False

    try:

        subprocess.run(['msconvert', os.path.join(directory, wiff_file), '-o', mzml_directory, '--mzML', '--64'])
        os.remove(temp_command_file)
        time.sleep(2)

        return True

    except FileNotFoundError:
        print("msconvert (Part of proteowizard) could not be found. Did you add the required directories to your system's PATH?\n")
        QApplication.processEvents()
        return False

    except subprocess.CalledProcessError as cpe:    
        print(f'Subprocess error converting {wiff_file} to mzML: {cpe}\nTraceback: {traceback.format_exc()}\n')
        QApplication.processEvents()
        return False

    except Exception as e: 
        print(f'Unexpected error converting {wiff_file} to mzML: {e}\nTraceback: {traceback.format_exc()}\n')
        QApplication.processEvents()
        return False
```

chambm, 2025-03-24:
Sorry for the delay responding. I don't understand the issue though. What does "the full filename of the first and last scans is truncated" mean? I the mzML output you shouldn't see full filenames for each scan.

ChristianIeritano, 2025-03-24:
Thanks for the response. Please run the Github module linked in my first post using the example data and whichever recent version of msconvert you folks are on and you will see what I mean by "truncated filenames". There is likely an error in the logic of the msconvert code, where the first and last scan from a .wiff file extracted and converted to .mzml does not have the correct name. Probably an index issue when looping, which I can't make a pull request for since I can't see the msconvert code. 

We know that this is an issue with the windows msconvert .exe as filename truncation is an issue for all versions >3.0.2, but not an issue for the archived version that we have (3.0.11392).

chambm, 2025-03-24:
Oh, you're talking about sample names within the wiff file. Not filenames. Because the wiff file _is_ the filename. I see what the issue is. Before 2020, wiff output filenames were always in the format `<filename>-<sample name>`. After that, I made it so that if the filename already contains the sample name verbatim, then it would forgo the sample name suffix. Because the sample names are very short in your file, e.g. "208" and "232" for the first and last samples, it's easy for the filename to contain those short strings. So those suffixes are excluded. With multiple samples, that's definitely a bug (both 208 and 232 suffixes are excluded, and so their output filenames are the same!). I think I will change that logic so it's only used for single-sample files.