ProteoWizard/pwiz #3175: Issue with demultiplexing Thermo Raw staggered windows

ZeshanAlimq, 2024-09-27:
Thanks for building this amazing platform.

I’m attempting to build a chromatogram library using gas-phase fractionation (GPF). I ran six DIA staggered GPFs covering the 400-1000 m/z range on the Thermo Eclipse.
I attempted to demultiplex using MSConvert on Windows using the settings shown in the image. [image]

I'm getting the following error. 
[image]

Raw files and method files are available here (file://SCI-6682/Users/45774862/OneDrive%20-%20Macquarie%20University/BrainRAW)
Thank you

nickshulman, 2024-09-27:
The link where you say to download the files does not work. I think that URL only makes sense on your own computer.
Maybe you could upload the files here:
https://skyline.ms/files.url

ZeshanAlimq, 2024-09-27:
Hi Nick,
Thanks for the quick response.
I have uploaded the files with the Zip name Zeshan.

nickshulman, 2024-09-27:
Thank you for uploading those files.
Your data does not seem to have any overlapping isolation windows.
The isolation window for every spectrum is [519.76501-528.76501]
You can use ProteoWizard SeeMS.exe to look at your data.
[image]

The "unknown exception" that MSConvert is hitting is "integer divide by zero". We should probably fix MSConvert so that it gives a better error message in this case.

ZeshanAlimq, 2024-09-27:
Thank you Nick for the help.