ProteoWizard/pwiz #3176: Precursor Mass Thermo RAW Files

lutfia95, 2024-09-27:
Hi, 

I am wondering, how are you correcting the mass of the precursor (m/z), I already checked the code part [(Link)](https://github.com/ProteoWizard/pwiz/blob/0939de1d732be8dadc8049def1c6ba1aa01e6b7f/pwiz_aux/msrc/utility/vendor_api/thermo/RawFile.cpp#L1813C1-L1838C10). 

The `monoMz ("Monoisotopic M/Z:")` is `929.9882`   
The precursor mass(`isolationMzPossiblyWithOffset`) from is: `930.3245239257812`  
Considering both, this will give a difference of  `0.3363239257812438`   
`iw ( "MS2 Isolation Width:")` is 1.6 
We don't use any offset so 0 and the `scanDescription ` is empty. How/where is the correction afterward is made? the resulted number from msconvert is something like  `929.9881......`

Thx! 
best

chambm, 2024-09-27:
If the "Monoisotopic M/Z:" value is within the isolation window, it is used directly. No correction is made. There are a couple filters you can try to adjust the precursor m/z but they are rarely useful these days: precursorRefine and precursorRecalculator. Both only available from command-line.

lutfia95, 2024-10-02:
thank you very much for your response! if the "Monoisotopic M/Z:" is not within the isolation window, is it corrected automatically? like the example above? I am interested in the correction method, if you could pin it to me in the code.

chambm, 2024-10-02:
It falls back to the isolation m/z (the filter m/z retrieved directly from the API so it has more decimal places).

lutfia95, 2024-10-05:
Sounds good thx! What is the name of the variable of isolation m/z in the api? is that one from the reaction(0)?

lutfia95, 2024-10-06:
If the isolation window not in the

closed 2024-10-06 by lutfia95