ProteoWizard/pwiz #3184: Missing Charge State when converting Sciex wiff to mgf file

jack-jiang01, 2024-10-07:
Dear sir,

Missing Charge State when i converting Sciex wiff to mgf file (zenoTOF 7600, ProteoWizard 3.0.24277.6d542ed\msconvert.exe). Can you help me, give me the work version of software or correct comman code?

rtool1, 2024-10-29:
Hello, I'm having the same problem, not having charge state converting SCIEX Wiff to mzML (zenoTOF 7600, ProteoWizard Version: 3.0.24280-6588539).

chambm, 2024-11-01:
Thanks for providing a test file. I'm able to view that with the current SeeMS. Perhaps you're using an older version? I'm also able to see charge state when opening the WIFF2 file. I'm not able to see charge state with the WIFF1 though. The WIFF1 API reports ParentChargeState is 0.

rtool1, 2024-11-04:
Thanks for taking a look at the 7600 test file, Matt.
I've downloaded the latest version on the ProteoWizard page, and I'm running in Windows 11.

I'm having trouble opening the wiff2 file with SeeMS (v3.0.24305.0); I can open the wiff file fine.  This seems like an easier way to diagnose.

The error seems to be "WiffFile2Impl:setSample() value cannot be null"
Could it be related to issue #2327 ?

[image]
[image]

chambm, 2024-11-04:
The D_20uL file? Weird, I'm able to open that wiff2 without a problem.

rtool1, 2024-11-04:
Ok, thanks!
MSConvert and SeeMS works if I run the apps as Administrator.

I think my issue's resolved.  Thanks!