ProteoWizard/pwiz #3186: `MSConvert` outputs wrong encoding for Waters raw files

t0mdavid-m, 2024-10-08:
We are using `MSConvert` to convert Waters `raw` files to `mzML`. Unfortunately we have been experiencing issues with downstream processing. The issues seem to be caused by the encoding of the `mzML` files.

The files are encoded with `Windows-1252` but in the header `UTF-8` is reported:
```
<?xml version="1.0" encoding="utf-8"?> 
```

This causes our XML-Parser to assume `UTF-8` and fail when running into non-ASCII characters (in our case `µ`/`b5` was problematic).

I would suggest to either convert the files to `UTF-8` or report the correct encoding in the header.

chambm, 2024-10-08:
What element/attribute had the non-ASCII character? Filepaths should be UTF-8 encoded and XML ids and idrefs should be _xHHHH_ encoded.

chambm, 2024-10-08:
Interesting. Looks like Visual C++'s isalpha() function says µ is an alphabetic character. But their example code implies it's not:

https://learn.microsoft.com/en-us/cpp/c-runtime-library/character-classification?view=msvc-170
> Generally these routines execute faster than tests you might write and should be favored over. For example, the following code executes slower than a call to isalpha(c):
> 
> if ((c >= 'A') && (c <= 'Z')) || ((c >= 'a') && (c <= 'z'))
>     return TRUE;

Looks like I need to go for the simpler approach.

PR #3204 (2024-10-25): - fixed encode_xml_id() allowing some non-alphabetical characters like µ
- fixed encode_xml_id() allowing some non-alphabetical characters like µ
- fixed msconvert --outfile not allowing Unicode in filename
- added IonMobility parameter to help text for titleMaker
Closes #3186 
Closes #3194
closed 2024-10-25 by chambm