ProteoWizard/pwiz #3209: Combined ion mobility format

SteffenHeu, 2024-10-31:
Hi MSConvert/Proteowizard team,

based on the request (https://github.com/mzmine/mzmine/issues/2169), I implemented the combined ion mobility format in mzmine.

When testing the results, i noticed that the combineIonMobility option fails to produce correct files as soon as peak picking is enabled.

I ran from the msconvert gui with the following command line:

```
"C:\Users\Steffen\AppData\Local\Apps\ProteoWizard 3.0.24255.d464da9 64-bit\msconvert.exe" --zlib --combineIonMobilitySpectra --filter "peakPicking vendor msLevel=1-" --filter "titleMaker <RunId>.<ScanNumber>.<ScanNumber>.<ChargeState> File:"""^<SourcePath^>""", NativeID:"""^<Id^>""""  "D:\OneDrive - mzio GmbH\mzio - shared\Example data\Waters\ims_mse_20180205_0099.raw"
```

The file: https://mzio-my.sharepoint.com/:f:/g/personal/ansgar_korf_mzio_io/EqIT9DiWBzpChMhAatVOvksBZoBj2Q2NUSt51_Nb0UI8Lg?e=ccU1TJ

The resulting file contained only 2 binary data arrays and a single ion mobility value in the scan header, corrupting the file. 
The log (Waters): 
```
D:\OneDrive - mzio GmbH\mzio - shared\Example data\Waters\ims_mse_20180205_0099.raw
------------------------------
Starting...
Opening file "D:\OneDrive - mzio GmbH\mzio - shared\Example data\Waters\ims_mse_20180205_0099.raw" for read...
Calculating SHA1 checksum...
Processing...
Writing "D:\OneDrive - mzio GmbH\mzio - shared\Example data\Waters\ims_mse_20180205_0099.mzML"...
[SpectrumList_Waters]: vendor centroiding is not supported for Waters ion mobility data

[SpectrumList_PeakPicker]: vendor centroiding requested but not available for non-vendor formats (and UNIFI); falling back to local maximum peak picker
```

The log (Agilent):

```
Writing "D:\OneDrive - mzio GmbH\mzio - shared\Example data\Agilent\LC-DTIMS-MS_6560_PFAS_MS1_only\merged\NISTSERUM.d.DeMP.mzML"...
[SpectrumList_PeakPicker]: vendor centroiding requested but not available for non-vendor formats (and UNIFI); falling back to local maximum peak picker
```

I tested with files from Waters and Agilent instruments, the results were identical. I can see the log message that the centroiding is not support for ims data, but it works as expected if the combine option is not used. 
I suspected the --combineIonMobilitySpectra being mentioned before the peak picking in the commandline being an issue, but it produced the same error.

1. I feel like the conversion should actually fail if the centroiding is not supported instead of running as normal and producing a file that does not contain the expected properties.
2. It would be great if the conversion was actually possible with peak picking enabled. From reading the log messages and knowing that peak picking works if the scans are not combined, I would assume that the scans are merged first and then the peakpicking is applied. 
    - Is it possible to change the order of these steps? 
    - Could the issue of the peak picking not being supported be solved by applying it to the mobility scans first and combining the scans afterward?

The reason we're asking for this is that we usually see that the vendor algorithms perform better on their own data than algorithims that are made to work on generic data types. Hence, we try to use vendor peak picking where possible.

Best,
Steffen

chambm, 2024-10-31:
The Waters and Agilent SDKs just don't support peak picking for IMS data either for combined or uncombined representation. The CWT peak picking in pwiz works ok for combined spectra but the mobility scans are generally too sparse to be individually peak picked then merged. Because CWT peak picking only considers the m/z and intensity dimensions, it drops the mobility dimension entirely.

I will work on improving the error message to fix the confusing "vendor centroiding requested but not available for non-vendor formats (and UNIFI)" part. Probably just by including Agilent and Waters IMS in the parentheses and removing the [SpectrumList_Waters] message.

I could also make a "use vendor algorithm or abort" option but it's not a high priority.

SteffenHeu, 2024-11-01:
Ok that's too bad. For some reason i thought that the vendor algorithms would support that, because i thought i did not see the error message in the past when converting. Re-trying it now shows the error though.

Do you have a recommendation for peak picking on profile mobility scan data then?

chambm, 2024-11-01:
You can try peak picking on the uncombined spectra with CWT and see how that goes for your data. It's definitely technically possible to merge into the combined representation after that, pwiz is just not set up to do it. Mainly we just don't recommend peak picking on Agilent and Waters IMS data: Skyline just extracts chromatograms from the profile spectra.

SteffenHeu, 2024-11-05:
Ok, thanks. I'll close the issue then 👍

closed 2024-11-05 by SteffenHeu