ProteoWizard/pwiz #3215: Wrong parentFile fileName for mzXML in Windows

JosuaCarl, 2024-11-11:
The mzXML files are saved with wrong file delimiters for the parent file, when using Windows. The conversion to mzML worked as expected with the correct delimiters.

With Windows:
```
<parentFile fileName="file:///D:\folder\tests\test_files/minimal_file.mzXML"
                fileType="processedData"
                fileSha1="4d590e5e85e6ae2eb812f076aa2cef64e0612be5"/>
```

With Linux:
```
<parentFile fileName="file:////mnt/d/folder/tests/test_files/minimal_file.mzXML"
                fileType="processedData"
                fileSha1=""/>
```

I encountered the error when testing, whether the correct input path is saved in the file.

chambm, 2025-03-24:
Which part is wrong? The final forward slash after \test_files? Because backslashes in Windows are correct (but forward slashes work too; mixed slashes can be a problem though).

JosuaCarl, 2025-03-24:
The problem is that it's mixed, making it hard to parse automatically. It starts with / and ends with \. See the `!` where I marked the slash.

Sidenote: Generating a `.mzML` file always uses the OS's slashing flavor.

PR #3414 (2025-04-25): Various fixes and Bruker selectedIon CCS
- fixed error handling reading signal chromatograms from Agilent data (reported by Gioele)
- fixed reading Bruker TSF data when only line or profile spectra are available (reported by Agnes)
- fixed mixed slashes in mzXML parentFile (reported by Josua)
- added CCS values for Bruker timsTOF selectedIons with known charge state

Closes #3215
closed 2025-04-25 by chambm