ProteoWizard/pwiz #3281: Parsing .wiff2 to stdout contains non-xml characters

SteffenHeu, 2024-12-17:
Hi,
i noticed that converting a SCIEX .wiff2 file to stdout with the option "-o -" will cause the output stream to contain the string `Could not find configuration for IPerformanceMeasurementEvent.` before the xml header starts.

I used the following command: `msconvert "path.wiff2" -o - --filter "peakPicking vendor msLevel=1-" --filter "scanNumber 1-1"`
(the scan number filter was only added so i could look at the output when i run msconvert from my cmd.)

Since there is also other output i'm guessing that everything else is sent to a logger/error stream and only the line  `Could not find configuration for IPerformanceMeasurementEvent.` is printed to the standard output stream (maybe using a printf or so).

.wiff2:
```
format: mzML
    m/z: Compression-Zlib, 64-bit
    intensity: Compression-Zlib, 32-bit
    rt: Compression-Zlib, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: -
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:
  peakPicking vendor msLevel=1-
  scanNumber 1-1

chromatogram list filters:

filenames:
  D:\Messdaten\mzio\sciex\2024_07_02_VW_018_LpMx_0-0100ug-mLEqui_POS_CID_30CE_top8_01.wiff2

processing file: D:\Messdaten\mzio\sciex\2024_07_02_VW_018_LpMx_0-0100ug-mLEqui_POS_CID_30CE_top8_01.wiff2
Could not find configuration for IPerformanceMeasurementEvent.
calculating source file checksums

writing output file: -\2024_07_02_VW_018_LpMx_0-0100ug-mLEqui_POS_CID_30CE_top8_01.mzML
<?xml version="1.0" encoding="utf-8"?>
```

.wiff:

```
format: mzML
    m/z: Compression-Zlib, 64-bit
    intensity: Compression-Zlib, 32-bit
    rt: Compression-Zlib, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: -
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:
  peakPicking vendor msLevel=1-
  scanNumber 1-10

chromatogram list filters:

filenames:
  D:\Messdaten\mzio\sciex\2024_07_02_VW_018_LpMx_0-0100ug-mLEqui_POS_CID_30CE_top8_01.wiff

processing file: D:\Messdaten\mzio\sciex\2024_07_02_VW_018_LpMx_0-0100ug-mLEqui_POS_CID_30CE_top8_01.wiff
calculating source file checksums

writing output file: -\2024_07_02_VW_018_LpMx_0-0100ug-mLEqui_POS_CID_30CE_top8_01.mzML
<?xml version="1.0" encoding="utf-8"?>
```

We are using the MSConvert -o - option internally in mzmine to parse some vendor formats and this currently breaks the import for .wiff2 files. I'll try to create a workaround internally in mzmine, but i think it would still be great if the issue was fixed.

Best

Steffen

chambm, 2024-12-17:
That output comes directly from the vendor DLL so we can't control what stream it goes to. I think there may be a way to suppress it with some extra configuration file that the API can read, but I'm not sure what incantation is necessary for that.

SteffenHeu, 2024-12-18:
Ok that's good to know, i thought it my be coming from msconvert. I have a workaround implemented, so you can close the issue unless you want to keep it as a backlog.