ProteoWizard/pwiz #3344: FR: SpectrumList_ScanSummer::sumSubScansNaive(): configurable m/z tolerance

rickhelmus, 2025-01-23:
Hello,

The scan summing option is sometimes recommended to convert TIMS-TOF data in a way it looks like 'regular' non-IMS LC-MS data. However, this currently doesn't work well when the spectra within an IMS frame are summed, since the current m/z tolerance is very low (<1E-6, see [here](https://github.com/ProteoWizard/pwiz/blob/d75b2a3d82cbc905f59e3ecab87e50147e6b3686/pwiz/analysis/spectrum_processing/SpectrumList_ScanSummer.cpp#L193)). The result is that mass peaks with close m/z are split. Would it be possible to make this value configurable to accomodate summing the relative 'noisy' TIMS MS spectra?

Thanks,
Rick

chambm, 2025-01-23:
You want to use a looser tolerance for binning? That binning step is actually intended to clean up the noisy summing of TIMS spectra IIRC. It takes place before the "summing" which takes into account the user-specified tolerances (precursor m/z, scan time, ion mobility of an uncombined mobility scan). Some illustrations might help me understand what you're seeing and what you want to see.

rickhelmus, 2025-01-23:
Hello,

Thanks for your fast response. To clarify, I am talking about MS (not MS/MS) data. Perhaps I misunderstood, but from what I could gather the current tolerances that can be configured (`precursorTol`, `ionMobilityTol` etc) are only used for the selection of PASEF MS/MS spectra that should be summed.

To illustrate the problem, the following three MS spectra of a compound (theoretical m/z 207.1492) are from the same sample analysis:

[image]
Raw data viewed with Bruker DataAnalysis: two separated m/z peaks: 207.1349/207.1493

[image]
Converted mzML file with combineIonMobilitySpectra and scan summing, plotted by SeeMS. Note that the splitting also reduces mass accuracy. Ideally these three peaks should be one peak.

[image]
Converted by [timsconvert](https://github.com/gtluu/timsconvert), plotted by SeeMS. Besides a bit of noise, there is one major peak at m/z 207.1492.

Thanks,
Rick

chambm, 2025-01-23:
Yeah, scanSumming isn't really the tool to use here. I think what you're asking is for is basically treating the noisy combined spectra as "profile" data and to centroid it. At one point I experimented with an option to treat combined TIMS data as profile data which could be peak picked with our CWT peak picker, but IIRC I wasn't too impressed with the results. I'm not sure if I had added "centroiding" to the CWT algorithm at that point though or if I did that later. If I did it later maybe that's why I wasn't impressed. I was probably also looking at MS2 data, not MS1. If you want to experiment with this, PRs are welcome. :) It should be fairly simple to have the combined IMS data be reported as profile so you can then apply peakPicking to it as you would for any pwiz profile data.

rickhelmus, 2025-01-30:
Ah, that is interesting... I hadn't really thought about centroiding the data. Thanks for your insights! I will see if I can find time to experiment with this.

Also related, I was digging a bit through the Bruker SDK and timsConvert source, and it seems the Bruker API already has a function to get centroided data: `tims_extract_centroided_spectrum_for_frame_v2()`. The API is a bit different though, so it will need some work to use it. Although it would be great to find a way that would also work for other IMS instruments (Agilent, Waters), because I guess there is no centroiding support for these either?

Thanks,
Rick

chambm, 2025-04-29:
The scanSumming filter doesn't care about the instrument, vendor, or centroid/profile. It only cares about whether the scans to be summed are within scan time and precursor m/z tolerance of each other (and if IMS is present, IMS tolerance). So if you can express what you want in terms of scan time and precursor m/z tolerance, it should work.

grandrea, 2025-04-30:
making it configurable would be great!