ProteoWizard/pwiz #3368: Using Msconvert in Proteowizard (Mac/Docker version)

davidtyus, 2025-02-12:
Hi everyone,

I am new to metabolomics and Docker (so hopefully the fix is simple) and trying to learn by reanalyzing some publicly available datasets. I found a suitable dataset but am stuck at trying to convert raw files to .mzml. I'm using a Mac so I'm trying to use msconvert through the "chambm/pwiz-skyline-i-agree-to-the-vendor-licenses" image on Docker. Though I successfully pulled the image all of the errors I'm getting suggest the image does not have msconvert (which I know it must!).

When I run this in the terminal there is no output or error: docker run --rm -v /path/to/my/data:/data --platform linux/amd64 chambm/pwiz-skyline-i-agree-to-the-vendor-licenses wine msconvert --help

And when I use "which msconvert" in bash I'm consistently getting a path not found error.

Has anyone come across this issue or have recommendations on what to try?

----

Also if it's helpful here's some more information about the files I'm trying to convert:

MetaboLights Dataset: https://www.ebi.ac.uk/metabolights/editor/MTBLS7807/files

The data was collected with MassLynx so the data is in a .raw folder with .inf, .DAT, .IDX, and .STS files inside.

Thank you for reading!

chambm, 2025-02-12:
Are you actually running on Mac? Intel chip, or Apple? The latter won't work without some kind of virtualization. I get this:

```
docker run --rm -v /path/to/my/data:/data --platform linux/amd64 chambm/pwiz-skyline-i-agree-to-the-vendor-licenses wine msconvert --help
Unable to find image 'chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest' locally
latest: Pulling from chambm/pwiz-skyline-i-agree-to-the-vendor-licenses
<snip>
Digest: sha256:ce423fab6312ee77425c32b348aecffd62a64c8dda334c8866a32aa626b62b6e
Status: Downloaded newer image for chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:latest
Usage: msconvert [options] [filemasks]
Convert mass spec data file formats.

Return value: # of failed files.

Options:
  -f [ --filelist ] arg              : specify text file containing filenames
  -o [ --outdir ] arg (=.)           : set output directory ('-' for stdout,
                                     '.' for current directory) [.]
  -c [ --config ] arg                : configuration file (optionName=value)
  --outfile arg                      : Override the name of output file.
<snip>
```

davidtyus, 2025-02-12:
Thanks for the quick response. Yes, I'm using Apple M1 chip. Sorry, I seem to misunderstand Docker and thought that running the container would serve as a sort of virtualization. 

And do you mean the instructions here are specific to the Intel Chip? https://hub.docker.com/r/proteowizard/pwiz-skyline-i-agree-to-the-vendor-licenses. (I also tried this image)

chambm, 2025-02-12:
Duplicate of https://github.com/ProteoWizard/container/issues/25

Yes, it would be great if we could run on Apple silicon, but I don't know how to make it happen or have hardware to test it. The vendor DLLs we use are compiled for AMD64, so some kind of virtualization is necessary and Docker doesn't handle it itself.

davidtyus, 2025-02-12:
Got it. Thank you for the explanation!

closed 2025-02-12 by davidtyus