ProteoWizard/pwiz #3370: msconvert creating two mzML files per raw file

jcleyle, 2025-02-13:
Hello,

I am using msconvert to convert Astral .raw files to mzML (from this publication: PMID: 38302753). For each raw file, msconvert is generating two mzML files. No matter which files I then try to analyze in Fragpipe, I am given the error message that some files (though not all) do not contain MS1. I have attached screenshots of the mzML settings. The mzML files themselves are too large to attach, but I can send them if it would be helpful.

[image]

[image]

Thanks!

chambm, 2025-02-13:
I tested on one of those files (20220909_EXPL8_Evo5_ZY_MixedSpecies_500ng_E5H50Y45_30SPD_DIA_1) and was not able to reproduce creating two mzML files. And indeed I can't imagine how MSConvertGUI could be doing that. I think that must be coming from Windows Explorer making copies of existing files and adding the `(2)` suffix on the end. The file I tested has MS1s and MS2s. Which file(s) did FragPipe complain about?

chambm, 2025-02-13:
Oh, BTW these are Orbitrap Exploris 480 files, not Astral.

jcleyle, 2025-02-13:
Ah yes sorry these are the Exploris files. One of the files it complained about is the 20220909_EXPL8_Evo5_ZY_MixedSpecies_500ng_E30H50Y20_30SPD_DIA_1_(2)_.mzML (I have tried running both sets of output files but am getting the same type of error).

chambm, 2025-02-13:
Did you really convert them without zlib? Maybe FragPipe has a problem with that? With your settings, I see both MS1s and MS2s in the output for 20220909_EXPL8_Evo5_ZY_MixedSpecies_500ng_E30H50Y20_30SPD_DIA_1.mzML.

jcleyle, 2025-02-14:
Yes I was converting without zlib. But after your comment about the Windows Explorer I ran msconvert from the command line - it fixed the double file output issue, and now Fragpipe is no longer throwing an error!

closed 2025-02-18 by chambm