ProteoWizard/pwiz #3387: Missing isolation window offsets when converting Waters DDA data

Gscorreia89, 2025-03-06:
Hi,

I have been using msconvert with the Waters DDA processing option to convert standard DDA experiments from Waters .raw to .mzL. 
I came across a specific error using [msPurity](https://www.bioconductor.org/packages/release/bioc/html/msPurity.html), and upon investigation, I noticed that the MSn/ ms level 2 scans are missing the parameters "isolation window lower offset" and "isolation window upper offset" in the  <isolationWindow> section. I managed to fix my specific problem by "patching" the mzML with a dummy value, but it would be good to have this values properly recorded in the mzML files. 

An example file, converted using msconvert and the Waters DDA processing option (centroiding with vendor first, then Waters DDA processing method) can be downloaded in this link - please let me know if you would like any more examples/details: 
https://drive.google.com/file/d/1QPIyTA5fXbBl5YnhoD1QFk6mKOtjRpD7/view?usp=sharing

Thanks,
Gonçalo

chambm, 2025-03-25:
Hi, I'd need the Waters .raw directory to look into this. The isolation window info just may be missing from the file so even with the ddaProcessing it doesn't know.

Gscorreia89, 2025-03-25:
Hi,

Please see this new link with the .raw and the matching .mzML: 
https://drive.google.com/drive/folders/1sL8EoKnVsY_gHMApUHY2P5ZaXd_lcT38?usp=sharing

These were acquired on a Waters Xevo G3, fastDDA method with an inclusion list. 
Thanks for looking into this, and please let me know if you need more details.