ProteoWizard/pwiz #3417: convert mzXML to mzML

Zyqiii1, 2025-04-03:
Dear ProteoWizard Team:
I am experiencing issues converting some mzXML files to mzML using ProteoWizard. DIANN can only use the mzML. While some files convert successfully, others fail with a red "failed" message. I have attached a sample of the mzXML files that fail to convert. Those converted mzML files can be identified by DIANN. Others cannot. If you need more information or the sample files, you can provide the email address and I will share the mzXML files. Thank you!

[image]

chambm, 2025-04-03:
If you select the failed row it should show why it failed in the bottom pane. Otherwise yes we'll need one of the failing mzXMLs to understand why it's failing.

Zyqiii1, 2025-04-04:
Thanks for your reply!
There are some examples for the wrong information:

**example 1:**
X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_119.mzXML
------------------------------
Starting...
Opening file "X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_119.mzXML" for read...
Failed - System.Exception: [SAXParser::ParserWrangler::elementEnd()] Illegal end tag "scan" at offset 5956907405.
   在 pwiz.CLI.msdata.ReaderList.read(String filename, MSDataList results, ReaderConfig config)
   在 MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   在 MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

**example 2:**
X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_127.mzXML
------------------------------
Starting...
Opening file "X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_127.mzXML" for read...
Calculating SHA1 checksum...
Processing...
Writing "X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_127.mzML"...
Failed - System.Exception: [SpectrumList_mzXML::spectrum()] Error seeking to <scan>.
   在 pwiz.CLI.msdata.MSDataFile.write(MSData msd, String filename, WriteConfig config, IterationListenerRegistry iterationListenerRegistry)
   在 MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   在 MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

**example 3:**
X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_129.mzXML
------------------------------
Starting...
Opening file "X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_129.mzXML" for read...
Failed - System.Exception: [SAXParser::ParserWrangler::elementEnd()] Illegal end tag "msRun" at offset 5572363363.
   在 pwiz.CLI.msdata.ReaderList.read(String filename, MSDataList results, ReaderConfig config)
   在 MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   在 MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

**example 4:**
X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_132.mzXML
------------------------------
Starting...
Opening file "X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_132.mzXML" for read...
Failed - System.Exception: [SAXParser::ParserWrangler::elementEnd()] Illegal end tag "index" at offset 5480708422.
   在 pwiz.CLI.msdata.ReaderList.read(String filename, MSDataList results, ReaderConfig config)
   在 MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   在 MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

**example 5:**
X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_155.mzXML
------------------------------
Starting...
Opening file "X:\zhangyuqi\ppi\PXD010288\mzXML\samon_I161214_155.mzXML" for read...
Failed - System.Exception: [SAXParser::ParserWrangler::elementEnd()] Illegal end tag "offset" at offset 2913599203.
   在 pwiz.CLI.msdata.ReaderList.read(String filename, MSDataList results, ReaderConfig config)
   在 MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   在 MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)

> If you select the failed row it should show why it failed in the bottom pane. Otherwise yes we'll need one of the failing mzXMLs to understand why it's failing.

Zyqiii1, 2025-04-04:
How can I revise the mzXML? Or what I can do is only to download mzXML again and convert it again. I have tried it for more than one time.

> If you select the failed row it should show why it failed in the bottom pane. Otherwise yes we'll need one of the failing mzXMLs to understand why it's failing.

nickshulman, 2025-04-04:
I am not sure, but those look like the sort of errors that you might get if the index in the mzXML file did not exactly match the contents of the mzXML file.
There is an optional `<indexOffset>` tag very close to the end of the mzXML file. If that tag is there, then it is supposed to indicate the exact byte offset within the file where there will be other tags which say exactly where to find each of the spectra.
There are some things that might happen to an mzXML file that would cause the value within that `<indexOffset>` tag to be incorrect.
One thing is that if you have used a text editor to edit the mzXML file, then those numbers would all end up being wrong.
Another thing that sometimes happens is that certain file transfer programs may change the line endings in the mzXML file which would also cause all the numbers to be wrong.
I am not sure exactly what you are supposed to do to fix this.
ChatGPT says to delete everything from `<index name="scan">` to `</indexOffset>`.

chambm, 2025-04-04:
Yeah, please share any of those example files and we should be able to figure out why it's happening quickly.

Zyqiii1, 2025-04-06:
Dear matt:

There are some files. I hope you can receive them.
My issue is https://github.com/ProteoWizard/pwiz/issues/3417

Best wishes,
yuqi

ZYQ 热爱！
***@***.***

        

         原始邮件

发件时间：2025年4月4日 22:59
收件人：ProteoWizard/pwiz ***@***.***&gt;
抄送：Zyqiii1 ***@***.***&gt;, Author ***@***.***&gt;

 	
 	 		 			从QQ邮箱发来的超大附件 	
 	 		 				 					 						 							 						 					
 					 						 							samon_I161214_127.mzXML (2).gz 							 (3.4GB, 2025年7月5日 17:0) 						 						 							进入下载页面 							：https://wx.mail.qq.com/ftn/download?func=3&k=9dc91939c9514205afba4d39676164349b9830ee656164344a190e5f4a03070700065a5f004c575052064e0d525751195a545a004807060301025109530505050028634a040c0b5a3d7f520f545355003d07510e4b0c1e6c2f7a431157484a5318225f9b8a40f6156d6f537ed8254c39d5de8b3099816534623663396561&key=9dc91939c9514205afba4d39676164349b9830ee656164344a190e5f4a03070700065a5f004c575052064e0d525751195a545a004807060301025109530505050028634a040c0b5a3d7f520f545355003d07510e4b0c1e6c2f7a431157484a5318225f9b8a40f6156d6f537ed8254c39d5de8b3099816534623663396561&code=b6c9ead4&from= 						
 					
 				
 			
 				 					 						 							 						 					
 					 						 							samon_I161214_129.mzXML (2).gz 							 (3.2GB, 2025年7月5日 17:4) 						 						 							进入下载页面 							：https://wx.mail.qq.com/ftn/download?func=3&k=99ca183798534302abb94c37366365330f696cf8346365334e1a0f511b02510304515001514e520a50004f035505011e5e0653551950015205505a0700015c04532b6244550e0a5d397c5301055154073904500e1a0e1f6b2b79421f064a4b541c218ef772e3751c279394b9e5cc9ae5fc5fad2ad6566733663562373463&key=99ca183798534302abb94c37366365330f696cf8346365334e1a0f511b02510304515001514e520a50004f035505011e5e0653551950015205505a0700015c04532b6244550e0a5d397c5301055154073904500e1a0e1f6b2b79421f064a4b541c218ef772e3751c279394b9e5cc9ae5fc5fad2ad6566733663562373463&code=f5b74ce3&from= 						
 					
 				
 			
 				 					 						 							 						 					
 					 						 							samon_I161214_132.mzXML (2).gz 							 (3.1GB, 2025年7月5日 17:8) 						 						 							进入下载页面 							：https://wx.mail.qq.com/ftn/download?func=3&k=cb994b66cd011350f9ea1f666331356172e30bae613135611c495c004e020c5902505703571c0c53075e1c525653504c0d04505e4c050d51505306005006015300783115005c5a0f6b2f0050500304556b5702544f5c4f39792a114e53181b064e723ffa5de4d27ef91f2597ab8b870cae2a3d19dbb33661346631666131&key=cb994b66cd011350f9ea1f666331356172e30bae613135611c495c004e020c5902505703571c0c53075e1c525653504c0d04505e4c050d51505306005006015300783115005c5a0f6b2f0050500304556b5702544f5c4f39792a114e53181b064e723ffa5de4d27ef91f2597ab8b870cae2a3d19dbb33661346631666131&code=4f1fa15a&from= 						
 					
 				
 			
 				 					 						 							 						 					
 					 						 							samon_I161214_155.mzXML.gz 							 (1.6GB, 2025年7月5日 17:13) 						 						 							进入下载页面 							：https://wx.mail.qq.com/ftn/download?func=3&k=9b9a17319e511657a9e94f31306130667936b154326130664c4a0c571d510251575d5952514c545e575d4c055752074b065103021f595150500150090402045f577f6142530c5f083b2c5007035301523b5454041c0c4a3e29294f56487573141c20e5f510bce4cc2d51ac8c0bc2fc0c1ec66531326130666465&key=9b9a17319e511657a9e94f31306130667936b154326130664c4a0c571d510251575d5952514c545e575d4c055752074b065103021f595150500150090402045f577f6142530c5f083b2c5007035301523b5454041c0c4a3e29294f56487573141c20e5f510bce4cc2d51ac8c0bc2fc0c1ec66531326130666465&code=dea12a0f&from= 						
 					
 				
 			
 				 					 						 							 						 					
 					 						 							samon_I161214_119.mzXML.gz 							 (3.6GB, 2025年7月5日 17:25) 						 						 							进入下载页面 							：https://wx.mail.qq.com/ftn/download?func=3&k=9bce446195514255a9bd1c613b61646409758089396164644c1e5f0716535d5055080156014c010507031f555c5853495c0702511451525d5550535100030151542b3212580c0b0a3b780357085355503b000358170c1e3c297d1c0643755680d3ad003a02bd33a80afeb4384133ebc41cfd3761396164646431&key=9bce446195514255a9bd1c613b61646409758089396164644c1e5f0716535d5055080156014c010507031f555c5853495c0702511451525d5550535100030151542b3212580c0b0a3b780357085355503b000358170c1e3c297d1c0643755680d3ad003a02bd33a80afeb4384133ebc41cfd3761396164646431&code=d12a9add&from=

Zyqiii1, 2025-04-06:
> Yeah, please share any of those example files and we should be able to figure out why it's happening quickly.

Can you download these files?

chambm, 2025-04-07:
As Nick suggested the problem is the indexOffset does not point at the actual index, probably due to the file being edited after the index was generated. ProteoWizard should really just disregard and rebuild the index in that case though, so that's an issue I'm going to fix. In the meantime you can delete the index at the end of the files (between `<index>` and `</indexOffset>`)

chambm, 2025-04-07:
OK, there's actually more than just that problem. Your files have some corrupted end elements. Like instead of `</peaks>` and `</scan>` sometimes it's `<Apeak>` or `<Cscan>`. I have no idea what could cause this. The GZIP files here fail their CRC check but still mostly decompressed, so I dunno if it's related to that.

closed 2025-04-09 by chambm