ProteoWizard/pwiz #3534: msconvert (docker image 3.0.25182-6ca4f8b linux/amd64) fails to handle volume-mapped symbolic links

edwardsnj, 2025-07-07:
The current msconvert docker image (3.0.25182-6ca4f8b (linux/amd64)) fails to handle symbolic links.

The previous image (3.0.25142-eb0d7d9) runs the commands below without issue.

Output for "-e WINEDEBUG=fixme-all+msgbox+relay" docker run option for current docker image:
[3.0.25182-6ca4f8b.log](https://github.com/user-attachments/files/21111015/3.0.25182-6ca4f8b.log)

Here is the command line (paraphrased) that has been working for quite some time, and that permits me to avoid copying the file to the local directory. I do not have direct write permission to the remotedisk. 

ln -s /mnt/remotedisk/internalfilename.raw propername.raw &&
chmod a+w . &&
docker run -v `pwd`:/data/ -v /mnt:/mnt chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:3.0.25182-6ca4f8b wine msconvert --filter 'peakPicking true 1-' --zlib --mzML --gzip /data/propername.raw

The current docker image and command-line *does* work if I first copy the raw file to the local directory.
The previous docker image and command-line *does* work through the symbolic link.

Thanks for any help you can provide. I will use the previous image for the time being.

edwardsnj, 2025-07-10:
I tested binding the file and that seems to work. Still need to bind `pwd` to /data to get the output file. Seems like a good workaround!

closed 2025-07-10 by edwardsnj

chambm, 2025-07-10:
Hah, that wasn't what I was suggesting, but it's even better. :)
I was thinking `-v /mnt/remotedisk:/data/` and doing something else (e.g. hardlink) to achieve the internalfilename.raw->propername.raw rename. I didn't even know you could bind mount a file, I thought it only worked for directories. TIL!

chambm, 2025-07-10:
If you keep track of performance metrics, I'd be interested to see whether you notice improvements in the new image. It _should_ be a pretty drastic improvement for Thermo RAWs.

edwardsnj, 2025-07-10:
Frankly, the msconvert step isn't the dominant cost in the pipeline. And since I can't run the symlink version with the current docker, I can't tell you if the speedup is due to the lack of symlink or just the new runtime or some other update to the msconvert code-base. I tested on a single RAW file using the command-lines above. Previous docker is about 3 mins using either approach. Current docker is about 1:30 mins, so it looks like about a 2-fold speedup. Pretty nice! 

Cheers!