ProteoWizard/pwiz #3566: timsTOF PASEF: "ms level" binary array (chromatogram data) doesn't match time and intensity arrays

btcooooper, 2025-07-25:
[Checked with latest version (3.0.25204-2ad5a87) after first seeing this problem with mzML files created by an early March version.]

Chromatogram data is stored by frame. Each (combined) MS1 spectrum takes up an entire frame, but frames with MS2 data usually have multiple spectra. TIC and BPC intensities should be sums over all spectra in each frame. But it appears that the "ms level" binary data is actually _per spectrum_ (which should be much longer than per frame), and simply truncated to the length of the time and intensity arrays. 

However it happened, the levels (1 or 2) in this array clearly do not line up with the time and intensity data.

btcooooper, 2025-07-28:
[image]

chambm, 2025-07-28:
There's an implied contract (on a best-effort basis) that the number of chromatogram TIC points should be equal to the number of spectra. So I'm trying to fix the PASEF chromatograms for the combined IMS mode to meet that contract. But there's also an implied constraint for chromatograms that time points should be unique. Are you sure that all the MS2 measurements are taken at exactly the same time? I know the accumulation is in parallel but I thought fragmentation is serial, so there should be some time separation between fragmentation events even if it isn't recorded in the TDF? Note that the time is technically "scan start time", not retention time.

btcooooper, 2025-07-29:
All spectra in a given frame have the same scan start time. I'm pretty sure that, in this context, "scan start time" means the start of the tims ramp and not the time at which ions were sent to the mass analyzer(s). That makes it very much like a retention time: it's consistently a fraction of a second later than the average ion accumulation time.

Confusingly, each MS2 spectrum is typically the sum of ~30 "scans," which in this context means individual TOF pushes that are co-added to improve SNR. Each tims ramp can have up to ~1000 TOF scans. A combined MS1 spectrum is the sum of all ~1000 TOF scans. The mobility written for an MS1 or MS2 spectrum is an average for all TOF scans included in the spectrum.

There are two options for fixing the inconsistency between the per-spectrum MS levels and the per-frame time and intensity arrays. One would be to write much longer, per-spectrum time and intensity arrays, and make sure the per-spectrum MS level array is the same length (not truncated). You'd need to interpolate between scan start times to give each spectrum its own time. And note that each MS2 spectrum in a given frame lists the same base peak intensity and the same total ion current—these must already be sums. So you'd also have to calculate these intensities for each MS2 spectrum separately.

But I think that it's reasonable to expect that a time array given as _chromatogram_ data should be interpreted as retention times. All TOF scans in a particular frame have the same tims "scan start time" and thus the same retention time. To me, it makes more sense to collapse the MS level data into a per-frame array.

chambm, 2025-07-31:
> You'd need to interpolate between scan start times to give each spectrum its own time. 
Yeah that's what I was talking about with an arbitrary time offset. If spectrum 1 (full frame MS1) is time 1, spectra 2-11 (grouped by DDA precursor or DIA window) are at time 2, and spectrum 12 (full frame MS1) is at time 3, then I'm adjusting spectra 2-11 at 1/10 intervals between 2 and 3 (e.g. 2, 2.1, 2.2, 2.3). Something like that.

btcooooper, 2025-08-01:
"// TIC and BPI will be repeated for each window in the MS2 frame, but there's no way to get accurate numbers without accessing binary data"

Annoyingly, the given per-frame TICs are not _exactly_ the same as the sum of TICs for all MS2 spectra in a frame (calculated from the binary intensity data). They're close, but typically off by a few counts.

PR #3572 (2025-08-06): - fixed chromatograms from timsTOF in combined IMS mode to match the …
- fixed chromatograms from timsTOF in combined IMS mode to match the number of spectra with the number of chromatogram points

Closes #3566
closed 2025-08-06 by chambm